opentrack/opentrack #815: Added pages to the Wiki

DanTDBV, 2018-07-14:
Based on you reply to Len62 here:
https://github.com/opentrack/opentrack/issues/810 
I have gone ahead and added the following pages to the Wiki:

https://github.com/opentrack/opentrack/wiki/opentrack-Shortcuts-descriptions

https://github.com/opentrack/opentrack/wiki/Manual-calibration-of-PointTracker-1.1-model

https://github.com/opentrack/opentrack/wiki/Loss-of-tracking-with-PointTracker-1.1-and-Cap-tip

I hope that they are helpful to some.

Len62, 2018-07-15:
Thanks for adding info to the wiki! I don't have the time at present but eventually I hope to get around to updating some items. I know for sure that sthalik will appreciate your efforts.

sthalik, 2018-07-15:
Thank you for the contributions!

I think credit for given page's principal author is ethical and in fact, a must. I'm leaving your attribution as-is. One thing I disliked about Wikipedia was being one of countless nameless editors who get no recognition.

I added some small elaborations on "Shortcuts" and "Loss of tracking" pages.

closed 2018-08-03 by sthalik