opentrack/opentrack #816: Not calibrating correctly in OT 2.3.10 when using PointTracker 1.1 and Cap.

DanTDBV, 2018-07-14:
Not really an issue, as I have solved it by doing a manual calibration instead.
Whenever I use your calibration method or the Len Endy method, the center + is offset to one side, as if the calibration assumes, that I am using a Clip instead of the cap.

It is consistent, as in every time I try. It could very well be me doing it wrong.

Here are a couple of pictures showing what I mean.

sthalik calibration method
[image]

Len Endy Calibration method 1
[image]

Len Endy Calibration method 2
[image]

I figured this out:
Directions are for moving the calibration marker (The +) in relation to you facing the screen, that is left is to your left and vice versa.

X 0 Y 0 Z 0 is the position of one of the LED’s (The LED with the + over it.)
(In my case, the top center LED of my cap.)
X + is to the right, - is to the left
Y + is above, - is below	
Z + is behind, - is in front

I set it up like this:
Manual Calibration
[image]

Now it works as a charm.

Once again, thank you so very much for this great software!

DanTDBV

sthalik, 2018-07-15:
The calibrator isn't assuming anything. It's generic statistical code. You can read it and not find a single reference to what the clip actually is. There's something else at play here. I don't know what, need to know more about camera position, camera rotation, etc.

sthalik, 2018-07-15:
As for the centering procedure I'd like you offer mine:

IMPORTANT: don't rotate too fast when calibrating. The calibrator only runs 30 times a second.

NOTE: don't roll on purpose. Just do natural movements as you'd do ingame. Roll calibration is a debug feature for users that can read and write code. 🐙 

NOTE: no need to yaw and pitch more than you'd do ingame.

- Pitch up as much as sensible
- `START`: yaw left/right
- Pitch up a bit lower (I do it with 3-5 pitch steps)
- If you're already pitched very low, go to `END`
- Pitch lower.
- Go to `START`
- `END`: Done, stop calibrating :octocat:

closed 2018-08-03 by sthalik