opentrack/opentrack #817: Warthog POV Hat only read as one button in 2.3.10

mcifone, 2018-07-15:
When using Opentrack 2.3.10 and the Thrustmaster Warthog throttle POV hat (on the front end of the right throttle handle), Opentrack reads it only as one button. In my setup, pushing the hat down re-centers and pulling it back pauses tracking. In version 2.3.10, down and up have the same function even though it shows Button 128 and Button 130 as the keys.

When attempting to rebind, Opentrack only reads it as button 128.

Reverting to 2.3.9 seems to have fixed the issue.

sthalik, 2018-08-03:
@mcifone I fixed it for opentrack `2.4.0`. You can use the build `opentrack-test-20180803_01.7z` from <<http://opentrack.misaki.pl>>.

There hopefully won't be any new bugs.

closed 2018-08-03 by sthalik

mcifone, 2018-08-03:
Not sure if it's actually a bug, but the keys look like they have been slightly changed? The previous key down on the POV was 130 and is now 132. Previously, they were all 128.

Not a big deal. Rebound the key and we're good to go.

Thanks for the quick fix!

sthalik, 2018-08-04:
It's meant to allow for binding particular POV directions as discrete keys. That was my original intention, and you seeing the same key for all 4 (or 8) directions was a bug.

sthalik, 2018-08-04:
And thanks for testing that out.