opentrack/opentrack #818: [Feat] Being able to flip the camera output

MathijsG, 2018-07-17:
I bought a new monitor, I thought one with a tiny bezel would be nice, but now I discover I block many top pixels already when attaching the PS3 Eye camera clip on it. So I'm looking for ways to mount my camera again, and one of it is to attach it to my shelve above me, upside down. However I couldn't find a way to flip the camera image, and I was wondering would this be easily possible in Opentrack? Any thoughts on this? I think I'm not the only one wanting it to mount upside down.

sthalik, 2018-07-17:
Hey,

First off if you mount it too high, the LEDs are gonna display as one giving you little pitch range.

But most importantly you have this PS3 Eye mounting plastic, right? You can put something under the part that goes on top of the display (pushing the "leg" upward), or shave off part of the frontmost "leg" with some basic tools.

MathijsG, 2018-07-17:
Yeah I use this:

[image]

[image]

And I thought about filing off the bottom, but then it will render the whole clip useless, because it will break off the 'connection between the two 'layers'. Howevever, I could try to raise the thing with something in between. Haven't thought of that yet :-). I hope there will be a sweetspot where I still have enough grip to attach to, yet not blocking pixels.

It's just a 6.9mm bezel. I will fiddle with it, when I'm at home.

503Kento, 2018-07-17:
Can’t you just invert the pitch axis?  Seems like opentrack won’t really care that the camera is inverted.

MathijsG, 2018-07-17:
I think I should invert every axis then, will try that as a second option. :)

MathijsG, 2018-07-17:
[image]

Thanks Sthalik for your idea :-).

closed 2018-07-17 by MathijsG

sthalik, 2018-07-17:
> Can’t you just invert the pitch axis? Seems like opentrack won’t really care that the camera is inverted.

It will. There are multiple nasty geometry problems involved.

I'd just tape cardboard/random trash between the leg and the tightener. It'll effectively make the leg shorter without cutting it off, without sacrificing stability.

sthalik, 2018-07-17:
Damn can you stand that blue light? I couldn't stand either of these. But there's a solution involving the round blanks covering the camera's eyes. And glue. 😆

MathijsG, 2018-07-17:
I do like the red one, for seeing if it's tracking. But I don't really mind it, it has been this way for years now :-P.

sthalik, 2018-07-17:
The driver handles the lights like crap. Killing or crashing opentrack makes it stay on. I had to glue over the red one too, since opentrack tends to crash during development 😃

MathijsG, 2018-07-17:
Here the blue one stays on after I shutdown my PC (due to USB still being powered), which lights up my whole room at night :-P.