opentrack/opentrack #819: War Thunder on Linux

CJPilot, 2018-07-22:
I was wondering if there was a way to get opentrack to work with the native version of War Thunder. I read in a old topic that there was no Freetrack support for native Linux games. Is that still true? I would prefer if I don't have to have a opentrack version for X-Plane and another opentrack for War Thunder.

sthalik, 2018-07-23:
Maybe via a virtual joystick on Linux?

CJPilot, 2018-07-23:
Do I have to udev something to make it work? I read that you need Read-Write permissions for the file /dev/input/uinput/ but when I start tracking, it say it can't find /dev/uinput/. I'm confused to what it's asking me to do.

sthalik, 2018-07-23:
You might have `/dev/uinput`. Symlink it and add yourself group permissions.

sthalik, 2018-07-23:
As long as `/dev/uinput` has a sensible group you can add yourself to the group and relog.

CJPilot, 2018-07-24:
I don't have a /dev/uinput folder. Should I create one?

sthalik, 2018-07-24:
# find /dev -name uinput

CJPilot, 2018-07-24:
I got it to see the "joystick" in X-Plane. Thank you so much @sthalik!

closed 2018-07-24 by CJPilot