opentrack/opentrack #822: Troubles calibrating unstable version with debian

Attigliuzzo, 2018-08-07:
I wrote to you for compiling issues with my debian testing and latest code (issue 821). With win 10 and opentrack 2.3.10 i have had calibrated quickly the tracker and the octopus follow my movements.
Now it is a nightmare under linux. I tried several times but the octopus doesn't follow my movements. The yaw and pitch aren't correct, the octopus only zooms.
I create a new ini file, setted few parameters and i've recorded a small video (attached [here](http://it.tinypic.com/m/k4cnyp/1)) without calibration.
I do yaw and pitch slow movements (no roll when possible). 
Is there a my mistake or other?
Thanks to devs for this "free" app
Regards

sthalik, 2018-08-11:
I can't view the attachment. Please use a different site.

Attigliuzzo, 2018-08-11:
It was in flv format, now it is [here](https://youtu.be/De3QVE3WBgk).

Attigliuzzo, 2018-08-15:
I tried to calibrate manually (20,-10,-25) moving the yellow cross above my face. In Flightgear it is usable but in certain positions the yellow cross jump from a position to another one ([video here](https://youtu.be/GCLYE1aKrKY)). 
It shouldn't happen because the cross must have a fixed positon in the space.

sthalik, 2018-08-16:
Have you enabled dynamic pose for that cap?

Attigliuzzo, 2018-08-17:
Yes, it is enable by default.

sthalik, 2018-08-28:
You may want to lower this to 15 ms: https://github.com/opentrack/opentrack/blob/unstable/tracker-pt/ftnoir_tracker_pt_dialog.cpp#L87

You may also want to cut this bunch by half: https://github.com/opentrack/opentrack/blob/unstable/cv/translation-calibrator.hpp#L45 (only the three static constexpr variables).

I can't think of Linux-specific issues with the calibrator. You can compile both opencv and opentrack with `-march=native -O3 -ffast-math` and nothing bad with happen. Even `-flto` works.

It's probably your calibration procedure that's somehow not right.

DanielKinsman, 2018-08-28:
I have noticed the "dynamic pose" checkbox setting doesn't seem to stick. I turn it off (I am using a clip, not a cap), but every time I open the point tracker settings it appears to be enabled again.

That said, calibration seems to work fine for me. I am on [my own a fork](https://github.com/DanielKinsman/opentrack/tree/snapcraft/snap) though.

Attigliuzzo, 2018-08-28:
@sthalik: i edited values as you suggested 15 ms  (in ftnoir_tracker_pt_dialog.cpp file) and with 1;0,5;1,5 [the standard values (2,5;1,5;3,5) in translation-calibrator.hpp file), and i compiled with "-march=native -O3 -ffast-math  -flto" values too. I continue to have calibration issue.
I use the debian repo opencv 2.3.2 instead latest one (3.4.1) could it be the problem?

> It's probably your calibration procedure that's somehow not right.

I move my head left, right, up and down it is not so difficult to do. At the same way i setted up under win10 the stable version.

@DanielKinsman: I can successfully enable and disable the dynamic pose checkbox, it is saved every time. If you can calibrate it, why cannot i do the same? i'm confused....

Attigliuzzo, 2018-08-29:
I compiled latest opencv and opentrack with all previous values and.....nothing.
Now i get, calibrating, a lot of sample (69 today, yesterday 379 wow)  but the octopussy never rotate left, right, up, down. It zooms or goes upside down when the cross jumps. 
After calibrating, the yellow cross is far from my center face and in some positions of my head it jumps, i can move manually but it is the same.

DanielKinsman, 2018-08-29:
I am also using opencv 3.2.0 so I don't think that is the problem. You could try building and installing my snap, then you should end up with the exact same binaries and libraries as me, as long as you can get snapd working on your distro.

sthalik, 2018-08-29:
@Attigliuzzo 

1. Remember to pitch AND yaw at the same time, i.e. yaw at different pitch positions.
1. The dynamic pose checkbox is easy pickings, add a missing `tie_setting` invocation.
1. It doesn't matter which opencv 2.4+ you use, as long as it compiles. Otherwise, opencv 2.4 build errors are simple to fix, if any.
1. It was only an example so you see how -ffast-math doesn't affect precision. You can use the system package.

I'm thinking of disabling roll calibration entirely, you can replace in `cv/translation-calibrator.cpp`

```
        return used_yaw_poses[yaw_k] == 1 ||
               used_pitch_poses[pitch_k] == 1 ||
               used_roll_poses[roll_k] == 1;
```

with

```
        return used_yaw_poses[yaw_k] == 1 ||
               used_pitch_poses[pitch_k] == 1;
```

I lost a week's worth of work due to bad xargs invocation (fml). Need to fix the center method first.

Attigliuzzo, 2018-08-29:
@DanielKinsman 
I think, i'm not a PC programmer: the source code + any compilers= same binaries or not??
I posted my binaries [here](https://github.com/opentrack/opentrack/wiki/Linux-binaries), may you try to use mine? Just to understand if it is my fault (led radius is 12 px, min 5 px and max is 50 px, camera? i set gain to 0) or there is a bug somewhere. Thanks

@sthalik 
1) i'll try but i've tried any movements (except a 360° rotation);
2) it is arab for me, please explain; i was wrong i cannot disable the dynamic pose checkbox, it is always on, i cannot disable it.
3-4) i've learned something new;

Yep, good idea. Roll view is useless with flight sims. I only need pitch, yaw and zoom.

Recompiled: i get an average of one hundred samples but after the calibration the only movement is zoom at the end upside-down (when the cross jumps). 
If i zoom the octopus zooms at the end it goes upside-down. 
I never get, with a full head movement, the "blacked-grayed" octopus. This is strange!!

> I lost a week's worth of work due to bad xargs invocation (fml). Need to fix the center method first.

Thanks for all your spare time spent with this project.

DanielKinsman, 2018-08-29:
I doubt we have the same binaries. We probably have different source code (for headers of libraries, e.g. qt5-dev header files), different compiler versions, different system libraries installed. Sorry I can't try your binaries as that would probably require me messing with my system libraries to get them working. You can try mine though because I've packaged it up as a snap so it bundles the libraries in with it rather than using the system ones.

I have [a PR for fixing the checkboxes](https://github.com/opentrack/opentrack/pull/825).

sthalik, 2018-08-29:
@DanielKinsman,

- I'm working on the checkboxes. A solution seems to be ripping out all `is_equal`-related code. I'm simplifying the code to prevent misuse of the "traits" type. There are no obvious regressions.
- You can join the team if you want. I'm impressed at you diving straight into the options subsystem, the most gory and complex part of the software's core logic.

Attigliuzzo, 2018-08-30:
@DanielKinsman 
It not simple as i thought. Yes, i could install snapd and try your package. 
In the store there is an old version, could you upload yours?? I read your snap project, use a similar name or post it here. Thank you

DanielKinsman, 2018-08-30:
I don't think I can upload to the snap store because it only works in devmode. In fact I've pretty much given up on making a "production" snap at least until snap officially moves to core18.

You should be able to build and run it yourself though using docker as described in [the readme](https://github.com/DanielKinsman/opentrack/blob/snapcraft/snap/README.md).

Attigliuzzo, 2018-08-31:
Load it in a cloud space. I tried the old "snap version" of opentrack from the store, there isn't the jumping cross issue, but i observed that the sw (older and newer versions) isn't able to traslate my head rotations in a rotation it converts them in a sliding. Perhaps it happens with caps and not with clips, who knows....

sthalik, 2018-10-07:
@Attigliuzzo can you elaborate on "opentrack from the store", please?

Attigliuzzo, 2018-10-07:
I'm away from home (my pc and IR cap are there), i could try on next month. 
Which store? Do you mean the ancient version on snap? or is there a newer one?

sthalik, 2018-10-08:
I'm halfway done with .deb files, they don't have dependencies. Other than that they build etc. See https://github.com/opentrack/debian.

Attigliuzzo, 2018-11-01:
Finally i can test your last code. I think that a video (calibration+movements) could be useful instead of tons of words, so it is [here](https://youtu.be/SSGO78oe2Ic).

sthalik, 2018-11-02:
You need to select "dynamic pose resolution". Right now opentrack selects the leftward of your points as the centermost.

Attigliuzzo, 2018-11-02:
i cannot find the option "dynamic pose resolution", where is it?
[image]

sthalik, 2018-11-02:
Below the FOV, you have it selected.

I'm unsure why your left point is detected as the centermost point. Are your cap model settings correct?

Attigliuzzo, 2018-11-02:
I re-checked the geometry of my cap, it seems ok. Before the calibration the cross is on the left led but after this process it is far away (it is visible in the video at 3:21 when i did great movements of my head).

sthalik, 2018-11-03:
It can't be on the left side. It must be on the center LED. That shows a problem even before calibration is started.

Maybe try moving farther away from the camera. Anyway calibration won't work if it gets bad model+pixel data.

Attigliuzzo, 2018-12-22:
I've recorded another video ([here](https://youtu.be/zlC1cDPWVM0)) with the latest code, now i'm farther from the camera. But when i press "calibrate" the cross moves over the left led. 
I always have a lot of samples, instead of previous versions this one looks better. 
Hope it helps, merry Christmas

sthalik, 2018-12-22:
It won't calibrate if it's not on the center LED. It won't track properly either.

closed 2018-12-22 by sthalik