opentrack/opentrack #824: Opentrack CSV data logging transform to image(vedio frames) position

tjinjin95, 2018-08-27:
I want to relate head movement data with image(vedio frames) position, I have head movement data per frame like this , how to find the position in image? or on spherical surface?
[image]
thanks

tjinjin95, 2018-08-28:
my expression not clearly before, I want use coordinate(yaw , pitch, roll) to find the position in video fram,  I made a figure in follow, maybe their some error. the field of view will change with coordinate(yaw , pitch, roll), where yaw in range of [-180,180], pitch in range of [-90,90]. I want know the relation between coordinate(yaw , pitch, roll) with field of view.
[image]
thanks  @sthalik

closed 2018-10-13 by tjinjin95

sthalik, 2018-10-13:
There will be rotation value differences if the physical FOV changes. The same odd behavior manifests when you move without rotating with regard to the camera plane.

As for changing the FOV setting without affecting the physical lenses, there shouldn't be any effect. The code is robust for vastly incorrect FOV settings.