opentrack/opentrack #827: Headphone tracking

peter-mason-dpm, 2018-09-21:
I have bought some noise cancelling headphones they are tremendous if not a little pricy and I am, therefore, not happy to attach the Dalan clip to them.
is there any possible to track the headphones or ideally my eyes?

Len62, 2018-09-21:
It is not possible to track the headphones, or your eyes. Opentrack does not support face-tracking. You could look into facetrack noir (http://facetracknoir.sourceforge.net/home/default.htm) or freetrack (http://www.free-track.net/english/), both can do face-tracking.

peter-mason-dpm, 2018-09-21:
A hat is not me though I thank you for your reply. I will look again at facetracknoir but I did not get the feeling it was able to offer me the solution that I am seeking.

sthalik, 2018-11-03:
Unfortunately not, but look into Aruco markers on this project's wiki. Head/eye tracking doesn't give the quality level other tracking methods provide.

closed 2018-11-03 by sthalik