opentrack/opentrack #828: Tutorials?

MrNobodyX3, 2018-10-05:
Is there any tutorials as to how to get this to work with games?

I'm using WMR Odyssey to track through Steam VR. It seems to be getting an input/output but none of my games react...

MathijsG, 2018-10-05:
Which games did you use? And what Output protocol did you use? Please we need more info :).

MrNobodyX3, 2018-10-05:
Any / all games I have like MC, Eco, Overwatch, R6

and I'm use the input of Steam VR and output of mouse emulation

sthalik, 2018-10-05:
Are you running opentrack as administrator? Either run both the games and opentrack as administrator, or nothing at all.

Also, opentrack doesn't require admin privileges.

MrNobodyX3, 2018-10-05:
Yes I've tried both, admin and not admin

MathijsG, 2018-10-08:
And the games DID work earlier with this? Many games don't use the direct Mouse input. I needed it to change to direct in GTA V to work. But I rather play real supported games with this :).

sthalik, 2018-11-03:
In theory it should work everywhere. Can you give an example of any game where it doesn't work so I can do further research?

MathijsG, 2018-11-03:
In GTA5 I needed to explicitly enable something like: Mouse Input: Direct

Will look up the name of it. The mouse input mode that was default didn't work with Opentrack

sthalik, 2018-11-03:
@MathijsG I understand, and have bought GTA5 a while ago.

sthalik, 2019-02-11:
I added a legacy input mode. It works but drifts away from center constantly.

closed 2019-02-11 by sthalik