opentrack/opentrack #831: Question: 'Straight' curves

Portash, 2018-10-07:
Is there a way to keep straight lines between the control points rather than a curve being automatically applied?

Portash, 2018-10-07:
Thanks. 

The reason I ask is that I'm getting a 'bump' right on the center-line of the axis - but it only happens if there's a curve applied. It's subtle but because it happens bang on the center line it's very annoying in use. 

If I remove the curve then the center 'bump' is gone, but then I no longer have a useful curve. It feels like one side of the axis is not starting from zero when a curve is applied.

sthalik, 2018-10-07:
You can add additional points to make it behave better. Best add at least 3-4 more points. It's also best if they're not overly distant from each other. That applies to all mappings, not just straight ones.

Portash, 2018-10-07:
OK, thanks. I'll do some experimenting.

Korny87, 2018-10-08:
Without starting deadzones on my curves everything is OK. Just be sure you pulled low point to 0,0.

sthalik, 2018-11-03:
@Portash what's your intended mapping?

Portash, 2018-11-03:
Sorry, I don't understand the question?

sthalik, 2018-11-03:
Can you draw something in Paint on top of the mapping window? What shape do you want?

Portash, 2018-11-04:
This is the kind of curve I used before the center 'bump' became too annoying for me....

https://imgur.com/65wBrNJ

I've also tried adding extra points on the curve but it didn't help.

sthalik, 2018-11-04:
But you wanted straight lines. What kind of straight lines do you want? opentrack doesn't support it but you can draw lines with mspaint.

Portash, 2018-11-04:
Basically the same curve but using straight lines (assuming it's the application of curves that causes the 'bump')

[img]https://i.imgur.com/uPBFP3d.png[/img]

sthalik, 2018-11-10:
TL;DR version: Is the attached image what you wanted? That "bump" in the middle depends on where you place points after Y=165.

The attached image is probably what you wanted. If not, see if adding extra points, and changing distances between them helps with the problem. In case this doesn't help, we'll figure something out.

[image]

The mapping takes into account (technical details):

- two previous points and the next one.
- the distance between different points matters, it also depends on how the spline is shaped overall.
- if there aren't enough points after some position, the last one is treated as if it repeated itself. it can produce strange results.

Basically we have several options on how to do mappings in the program:

- always linear
- current version (Catmull-Rom)
- cubic (it doesn't go right through the control points)

opentrack could also offer all of these options. But I think the app needs to have some sort of coherency, a permeating idea. This is why there aren't too many options for mapping–just like for other options.

Portash, 2018-11-20:
Thanks for the guidance but the 'bump' is still there. I've tried many variations of curve with different amounts of points but haven't been able to dial the bump out.

Not to worry, I'll continue using a straight line - it's not a big problem and I've more or less got used to it.

closed 2018-11-20 by sthalik