opentrack/opentrack #833: problem for log data when head movement from yaw =-180 to yaw =180 degree

tjinjin95, 2018-10-13:
I have sampled some log data of videos, but I found when head movement from from yaw =-180 to yaw =180 degree have a bias angle . the log data like the follow, the  part where I inserted a quote are the skip positions. I calculated the rotation angle of the two skips are 12.5768 (radian) and  26.6360  (radian). The two angles are much larger than other angles.

     Yaw       pitch      roll
-176.561	-2.218	-12.486
-179.015	-3.313	-12.441

>  -179.015	-3.313	-12.441
>  177.678	8.773	9.593

177.678	8.773	9.593
175.394	9.995	9.902
175.394	9.995	9.902
173.017	11.419	12.064
173.017	11.419	12.064
170.614	12.835	15.196
170.614	12.835	15.196
168.634	13.575	16.906
168.634	13.575	16.906
167.151	13.456	17.735
167.151	13.456	17.735
166.067	12.447	19.041
166.067	12.447	19.041
165.744	10.695	21.252
165.744	10.695	21.252
165.676	8.893	22.383
165.676	8.893	22.383
165.829	7.653	20.448
165.829	7.653	20.448
166.219	7.292	13.784
166.219	7.292	13.784
166.741	6.966	4.63
166.741	6.966	4.63
167.356	6.561	-3.676
167.356	6.561	-3.676
168.147	6.734	-7.331
168.147	6.734	-7.331
168.84	6.821	-10.926
168.84	6.821	-10.926
169.092	6.388	-16.143
169.092	6.388	-16.143
169.731	5.76	-20.862
169.731	5.76	-20.862
171.101	5.055	-24.953
171.101	5.055	-24.953
173.02	5.068	-27.662
173.02	5.068	-27.662
174.747	5.222	-30.69
174.747	5.222	-30.69
176.298	5.344	-35.009

> 176.298	5.344	-35.009
> -173.614	-19.298	35.554

-173.614	-19.298	35.554

sthalik, 2018-10-13:
Have you performed any centering, or used the "center on startup" option? Try turning that off.

tjinjin95, 2018-11-04:
what you said is right and I use quat2angle function from matlab Aerospace Toobox , which could Convert quaternion to rotation angles. 

for example ,
[yaw ,pitch ,roll] = quat2angle(quaternion, 'ZYX');
then I can get [yaw ,pitch ,roll].

I  find a phenomena is that when  yaw =-180 to yaw =180 degree and  yaw =180 to yaw =-180 degree,  pitch data will have a jump and make it discontinuous.