opentrack/opentrack #839: Unknown cmake command "include_guard"

hoover67, 2018-11-02:
Trying to configure opentrack via 

git clone ....
cd opentrack
mkdir build
cd build 
ccmake ..

and trying to configure the options using "c" I'm getting the following error: 

Make Error at cmake/opentrack-mrproper.cmake:10 (include_guard):
   Unknown CMake command "include_guard".
 Call Stack (most recent call first):
   CMakeLists.txt:32 (include)

Linux Mint 18.2. 

Thanks Uwe

sthalik, 2018-11-02:
What's your CMake version? Remove the lines and nothing bad will happen.

closed 2018-11-02 by sthalik

hoover67, 2018-11-02:
Mint 18 comes with 3.5, but I installed a newer version using pip:

~/src/opentrack/build $ cmake --version
cmake version 3.12.0

the install_guard error is gone now (thanks for the tip).

ccmake still uses version 3.5 however, how can I tell it to use the newer version instead? I removed the old binary and created a symlink to /usr/local/bin/cmake, but to no avail.

Thanks! 

Uwe

sthalik, 2018-11-02:
Since this is a `.deb`-based system, I'd build it from the distro's sources. It's actually pretty unusual for `ccmake` to be of a different version.

Go for `aptitude download cmake`, followed by `aptitude build-dep cmake-curses-gui && aptitude build-dep cmake`, `dpkg-buildpackage` in the package's directory then install the required bits, which *should* include `cmake-gui`, `cmake`, and `cmake-curses-gui` all in separate `.deb` packages.

sthalik, 2018-11-03:
Given some thought, something bad is going with your `cmake` packages. `which cmake` and `which ccmake` should correspond to the same directory. If not, remove the non-standard package. When in doubt, reinstall `cmake-curses-gui` and `cmake`.

hoover67, 2018-11-04:
Hello Stanislaw,

thanks for the support & suggestions. What is your reference Linux build
system?

All the best, Uwe