opentrack/opentrack #841: Still cant get simconnect to work.

thewoolysheep08, 2018-11-03:
I still cant get simconnect to work, im confused with the instructions and i wish you wouldnt close my issue without asking if i could get it working...

sthalik, 2018-11-03:
You can respond in the original issue. I can reopen it. Just copy the file and name it like the original but without "lib" up front. Don't change the extension.

closed 2018-11-03 by sthalik

sthalik, 2018-11-03:
Go to the "modules" directory where you installed opentrack. Don't rename, __copy__ it.

Copy `libopentrack-proto-simconnect.dll` to `opentrack-proto-simconnect.dll`. Notice the lack of the "lib" prefix on the copied file.

thewoolysheep08, 2018-11-03:
rename libopentrack to just opentrack?

sthalik, 2018-11-03:
DO NOT RENAME BUT COPY! 👍 

Do that for that simconnect dll file. Leave rest as-is.

thewoolysheep08, 2018-11-03:
right, i got it! thanks so much, i apologize  for how long it took when it comes to stuff like this I'm a complete noob