opentrack/opentrack #842: build error on Linux Mint 18.2 (64bit)

hoover67, 2018-11-04:
I've now installed a recent cmake version from source (3.13.x), looks like both ccmake and cmake are happy now about the version. 

After configuring the install path and the SDK_WINE_PREFIX and trying to start the build, I get the following error: 
hoover@feynman ~/src/opentrack/build $ cmake ..
-- User settings file 'sdk-paths-hoover@GNU-Linux.cmake' doesn't exist
-- module executable
-- module api
-- module compat
-- module csv
-- module cv
-- module filter-accela
-- module filter-ewma
-- module user-interface
-- module logic
-- module migration
-- module options
-- module pose-widget
-- module proto-fgfs
-- module proto-libevdev
-- module proto-udp
-- module proto-wine
-- module qxt-mini
-- module spline
-- module tracker-freepie-udp
-- module tracker-fusion
-- module tracker-hatire
-- module tracker-pt-base
-- module tracker-pt
-- module tracker-s2bot
-- module tracker-test
-- module tracker-udp
-- -- Compile flags:
--  CXX: -fvisibility=hidden -fuse-cxa-atexit 
--  CXX_RELEASE: -O3 -DNDEBUG -UNDEBUG
-- -- Link flags:
--  LINK_SHARED: -Wl,-z,relro,-z,now,--exclude-libs,ALL 
--  LINK_SHARED_RELEASE: 
-- -- Static archive flags:
--  STATIC: 
--  STATIC_RELEASE: 
-- --
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hoover/src/opentrack/build

hoover@feynman ~/src/opentrack/build $ make 
[  0%] Generating moc_meta.cpp
/home/hoover/src/opentrack/compat/meta.hpp:10: Parse error at "meta"
compat/CMakeFiles/opentrack-compat.dir/build.make:117: recipe for target 'compat/moc_meta.cpp' failed
make[2]: *** [compat/moc_meta.cpp] Error 1
CMakeFiles/Makefile2:1483: recipe for target 'compat/CMakeFiles/opentrack-compat.dir/all' failed
make[1]: *** [compat/CMakeFiles/opentrack-compat.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

The named hpp file looks harmless enough, but then again I'm no c++ expert.

hoover67, 2018-11-05:
> What's your cmake version? Right in opentrack-platform.cmake we have:
>
> set(CMAKE_CXX_STANDARD 17)
> set(CMAKE_CXX_STANDARD_DEFAULT 17)
> set(CMAKE_CXX_STANDARD_REQUIRED TRUE)
> set(CMAKE_CXX_EXTENSIONS FALSE)
>
> Which should do the trick.

Hello Stanislaw,

I'm using cmake 3.13.xxx (binarly tar.gz-file downloaded from the company's
website)

Can you recommend a specific opentrack version for compilation or should I
always go for the latest git repo? I read somewhere the the --unstable
branch is also available, which one should I be using for my attempts to
build a Linux binary?

All the best,

Uwe

hoover67, 2018-11-05:
I've gotten a bit further by setting up a clean Ubuntu 16.04 test environment (VM) and installing a ton of development packages (both their 64- and 32 bit versions. ccmake (13.3) still complains about not being able to find opencv (this is opentrack 2.3.9 from the "releases" section: 

grep -i opencv CMakeCache.txt
//The directory containing a CMake configuration file for OpenCV.
OpenCV_DIR:PATH=OpenCV_DIR-NOTFOUND
//MODIFIED property for variable: OpenCV_DIR
OpenCV_DIR-MODIFIED:INTERNAL=ON

But: 

# ls -l /usr/share/OpenCV/
total 48
drwxr-xr-x 2 root root  4096 Nov  5 11:40 java
-rw-r--r-- 1 root root 13535 Sep 13 16:04 OpenCVConfig.cmake
-rw-r--r-- 1 root root   378 Sep 13 16:04 OpenCVConfig-version.cmake
-rw-r--r-- 1 root root  4675 Sep 13 16:04 OpenCVModules.cmake
-rw-r--r-- 1 root root 12573 Sep 13 18:06 OpenCVModules-release.cmake

and: 

# dpkg -l libopencv-dev
ii  libopencv-dev                      2.4.9.1+dfsg-1.5ubuntu amd64                  development files for opencv

I tried setting the path in ccmake to /usr/share/OpenCV, but it keeps complaining about the missing file(s). 

Can anyone name a reference platform (Linux) on which building opentrack from source works (with a reasonable amount of jumping through hoops, I've compiled many a kernel in my 20 years on Linux but opentrack has me on my knees for the time being :) 

Also when building the wine wrapper, winegcc cannot find the correct libc++ (32bit I assume); OTOH, I cannot install wine ia32 without pulling in a ton of 32bit bit dependencies which will probably break the system if I pull all of those in. 

Thanks in advance for any pointers, 

Uwe

sthalik, 2018-11-05:
@hoover67,

> ccmake (13.3) still complains about not being able to find opencv
> [...]
> this is opentrack 2.3.9 from the "releases" section

opentrack 2.3.9 has an unnecessary requirement of opencv 3.0. Use the unstable build, it should be in good shape. It also has a great amount of improvements to keybinding support on X11.

> Also when building the wine wrapper, winegcc cannot find the correct libc++

On Debian I had to install packages with names beginning with `lib32`, and `libc6-dev-i386`.

sthalik, 2018-11-05:
> Can anyone name a reference platform (Linux) on which building opentrack from source works

It works on my Debian sid and Arch.

sthalik, 2018-11-05:
opentrack is a 64-bit application on amd64 linux. It's the wine wrapper that runs as a 32-bit process. It's because few people use 64-bit wine prefixes. They tend to break games.

> Does your debian test build support the wine wrapper / DLLs for use of opentrack with, say Falcon BMS running in Wine?

Yes. The wrapper is only for transferring data between Wine world and Unix world.

sthalik, 2018-11-05:
What do you want `dpkg -l` for?

hoover67, 2018-11-05:
Hello Stanislaw,

thanks for the reply. dpkg -l would be very helpful in determining if I
have installed all needed development packages required to run a successful
build. Debian Sid is quite ancient though, right?

All the best, Uwe

sthalik, 2018-11-05:
@hoover67 the package names vary between Debian and Ubuntu. Use the `apt-file` utility for finding what you need. I have `libc6-dev-i386:amd64` but on your end it may be named differently.

hoover67, 2018-11-06:
Thanks Stanislaw.

I'm now using the git version of the source and set up a fresh Ubuntu 16.04
install with cmake 3.12.x. I've automated the setup process to I can fire
up a throwaway build VM (in openstack) at a moment's notice, it
automatically installs all the development packages & tools and clones the
git repo.

Cmake appears to find everything needed, so I press "generate", then do
"cmake .." from the build directory followed by "make".

Now I'm still getting the "meta"-related syntax error:

-- stuff snipped --
[  2%] Generating moc_math-imports.cpp
[  2%] Generating moc_math.cpp
[  2%] Generating moc_meta.cpp
/usr/local/src/opentrack/compat/meta.hpp:10: Parse error at "meta"
compat/CMakeFiles/opentrack-compat.dir/build.make:117: recipe for target
'compat/moc_meta.cpp' failed
make[2]: *** [compat/moc_meta.cpp] Error 1
CMakeFiles/Makefile2:1473: recipe for target
'compat/CMakeFiles/opentrack-compat.dir/all' failed
make[1]: *** [compat/CMakeFiles/opentrack-compat.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

gcc / c++ version is
gcc (Ubuntu 5.4.0-6ubuntu1~16.04.10) 5.4.0 20160609

All the best, Uwe

sthalik, 2018-11-06:
Try in that order:

1. Go to build directory. Run `make mrproper ; cmake .`. Try building.
2. Edit `CMAKE_CXX_FLAGS` and add `-std=c++17` manually.

That GCC version is pretty ancient and I can't guarantee it can compile with the C++17 standard. But it's CMake's fault, it's supposed to add the C++ standard's version as per my earlier post. It doesn't.

hoover67, 2018-11-06:
Thank you, Stanislaw. 

I also sent you a private message on the BMS forum as to not clutter up this thread unneccessarily w/r to automating the setup of the build environment. 

I'd guess the gcc version on Debian Sid is *much* older than the one on xenial unless your debian testing environment it built on the latest "buster" release.  

I tried the steps above but still no joy (did I use the correct option syntax?): 

snip snip.... 

--  CXX: -fvisibility=hidden -fuse-cxa-atexit -std=c++17
--  CXX_RELEASE: -O3 -DNDEBUG -UNDEBUG
-- -- Link flags:
--  LINK_SHARED: -Wl,-z,relro,-z,now,--exclude-libs,ALL 
--  LINK_SHARED_RELEASE: 
-- -- Static archive flags:
--  STATIC: 
--  STATIC_RELEASE: 
-- --
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/local/src/opentrack/build
ubuntu@test-scue18:/usr/local/src/opentrack/build$ make 
[  0%] Generating moc_meta.cpp
/usr/local/src/opentrack/compat/meta.hpp:10: Parse error at "meta"
compat/CMakeFiles/opentrack-compat.dir/build.make:117: recipe for target 'compat/moc_meta.cpp' failed

sthalik, 2018-11-06:
What's your Qt version? I see this happens when the metaobject compiler is run.

hoover67, 2018-11-06:
Hello Stanislaw,

I've now installed the latest gcc-7 / c++-7 from cosmic-backports and added
the -std-Variable, also I've pointed cc make to the new "-7" executables
(gcc-7 ..ar-7, ... ranlib-7) and so on but I'm still getting the "meta"
compile error as mentioned above.

I assume using gcc-7 is necessary for building opentrack then? CMake still
doesn't pick up the "-dstd..." flag automatically though even with the "-7"
packages present on the system.

All the best, Uwe

hoover67, 2018-11-06:
Qt's at version 5.5.1.

All the best, Uwe

sthalik, 2018-11-06:
It's the Qt's metaobject compiler that fails to pick up `-std=c++17`. You can verify by running `make VERBOSE=1`. Try running `debootstrap` to initialize a prefix with some `.deb`-based newer version, e.g. Debian sid or maybe Ubuntu yakkety. You won't have problems with default package versions.

hoover67, 2018-11-07:
I've gotten a bit farther along using Ubuntu bionic as the build platform.
The compile works ok up to here:

[ 69%] Built target i18n-lang-stub-ts-proto-udp
[ 69%] Built target i18n-lang-zh_CN-ts-proto-udp
[ 70%] Generating opentrack-wrapper-wine.exe.so
In file included from
/usr/local/src/opentrack/proto-wine/opentrack-wrapper-wine-main.cxx:1:0:
/usr/include/c++/7/cerrno:41:10: fatal error: bits/c++config.h: No such
file or directory
 #include <bits/c++config.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
winegcc: g++ failed

apt-file shows numerous occurences of this header file, which one do I need
to install?

Right now, the following wine packages are installed on the system:

ii  fonts-wine                           3.0-1ubuntu1
 all          Windows API implementation - fonts
ii  libwine:amd64                        3.0-1ubuntu1
 amd64        Windows API implementation - library
ii  libwine-dev:amd64                    3.0-1ubuntu1
 amd64        Windows API implementation - development files
ii  libwine-development:amd64            3.6-1
amd64        Windows API implementation - library
ii  libwine-development-dev:amd64        3.6-1
amd64        Windows API implementation - development files
ii  wine-development                     3.6-1
all          Windows API implementation - standard suite
ii  wine-stable                          3.0-1ubuntu1
 all          Windows API implementation - standard suite
ii  wine64                               3.0-1ubuntu1
 amd64        Windows API implementation - 64-bit binary loader
ii  wine64-development                   3.6-1
amd64        Windows API implementation - 64-bit binary loader
ii  wine64-development-tools             3.6-1
amd64        Windows API implementation - 64-bit developer tools
ii  wine64-tools                         3.0-1ubuntu1
 amd64        Windows API implementation - 64-bit developer tools

All the best, Uwe

sthalik, 2018-11-07:
Great! Use `apt-file find` for that.

hoover67, 2018-11-07:
That's my problem, thanks :)

Apt-file finds about 70 files with that name excluding the more obvious
architechtures, I just have no idea which one winegcc uses and / or where
it expects it:

$ apt-file search bits/c++config.h | egrep -v
'arm|mips|powerpc|s390|sparc|alpha' | wc -l
68
I tried the subdirectory the winegcc error references:

$ apt-file search /usr/include/c++/7/bits/c++config.h

but the search comes up empty (as opposed to the nearly 70 matches for the
bits/... name above).

Thanks & all the best, Uwe

hoover67, 2018-11-07:
Small update: I installed the package

lib32stdc++-7-dev

and now the header file is found by winegcc. 

During the linking stage I now get the following error: 

[ 70%] Generating opentrack-wrapper-wine.exe.so
ld: Relocatable linking with relocations from format elf64-x86-64 (/usr/lib/x86_64-linux-gnu/wine/libwinecrt0.a(exe_entry.o)) to format elf32-i386 (opentrack-wrapper-wine.4cnv8F.o) is not supported

All the best & thanks in advance for any ideas! 

Uwe

sthalik, 2018-11-07:
Install purely 32-bit Wine.

sthalik, 2018-11-07:
What I did was install manually-compiled Wine into `/opt/wine-version` and symlink `/opt/wine-version/bin/*` into `/usr/bin/`. Then `winegcc` worked.

hoover67, 2018-11-07:
I found a 32bit package from winehq (stable, including the development stuff) and I'll do some more testing tomorrow, so thanks for the advice.

Just to be clear as I'm planning to use opentrack primarily with X-Plane (native Linux version) and Falcon BMS running in wine: Will I be able to use the wine wrapper and the DLLs with a 64 bit wine version? I need to run BMS in 64 bit so it's possible to join online servers that run the 64 bit binary. 

All the best, Uwe

sthalik, 2018-11-08:
> Will I be able to use the wine wrapper and the DLLs with a 64 bit wine version? I need to run BMS in 64 bit

Yes.

hoover67, 2018-11-08:
That's great news, Stanislaw, thanks.

As the german saying goes: "Mühsam ernährt sich das Eichhörnchen" :) 

Looks like most everything is compiling now after a bit of fiddling with the winehq repos (no worries, I'll share my build files once I'm certain the build works in a reproducible manner). I had a bit of trouble with the xplane plugin (see below). 

My package installs to 

/usr/local/openstack-git 

and hopefully works on all Ubuntu 18.04-based systems, let me know if you're interested in taking a look. 

The xplane plugin complains during the first run, then the build works on a 2nd make: 

[100%] Built target i18n
Scanning dependencies of target opentrack-xplane-plugin
[100%] Building C object x-plane-plugin/CMakeFiles/opentrack-xplane-plugin.dir/plugin.c.o
/usr/local/src/opentrack/x-plane-plugin/plugin.c: In function ‘XPluginStart’:
/usr/local/src/opentrack/x-plane-plugin/plugin.c:192:16: warning: passing argument 1 of ‘memset’ discards ‘volatile’ qualifier from pointer target type [-Wdiscarded-qualifiers]
         memset(shm_posix, 0, sizeof(WineSHM));
                ^~~~~~~~~
In file included from /usr/include/features.h:424:0,
                 from /usr/include/x86_64-linux-gnu/bits/libc-header-start.h:33,
                 from /usr/include/stdlib.h:25,
                 from /usr/local/src/opentrack/x-plane-plugin/plugin.c:9:
/usr/include/x86_64-linux-gnu/bits/string_fortified.h:59:1: note: expected ‘void *’ but argument is of type ‘volatile WineSHM * {aka volatile struct WineSHM *}’
 __NTH (memset (void *__dest, int __ch, size_t __len))
 ^
/usr/local/src/opentrack/x-plane-plugin/plugin.c: In function ‘shm_wrapper_init’:
/usr/local/src/opentrack/x-plane-plugin/plugin.c:83:5: warning: ignoring return value of ‘ftruncate’, declared with attribute warn_unused_result [-Wunused-result]
     (void) ftruncate(self->fd, mapSize);

Run make again: 

[ 98%] Built target i18n-lupdate
[100%] Built target i18n-qm-stub
[100%] Built target i18n-qm-nl_NL
[100%] Built target i18n-qm-ru_RU
[100%] Built target i18n-lrelease
[100%] Built target i18n
[100%] Built target opentrack-xplane-plugin

All the best, Uwe

sthalik, 2018-11-29:
@hoover67 how are are opentrack and linuxtrack going out for you?

hoover67, 2018-11-29:
Hello Stanislaw,

opentrack is working fine now, I replaced linuxtrack with my opentrack
build on Linux / X-plane now too and it works wonderfully.

I'm planning to release a new binary build in the beginning of September.

All the best, Uwe

sthalik, 2018-11-29:
I'm happy to hear this, especially the X-Plane plugin that went untested for a time.

closed 2018-11-29 by sthalik