opentrack/opentrack #843: 360 degrees tracking? + 180 degrees limit issue

alex-k8, 2018-11-04:
Hello, I'm using an arduino with an mpu6050 to track the rotation of my head, but from what I can see the input limit for yaw is 180 degrees.

Would it be possible to do full 360 degrees tracking somehow?

I tried it through my arduino code (for example if the degrees are 200, it would send -160, if they are 220, it would send -140 etc.), but that doesn't quite work well, firstly because I can't use a filter, which I need for smoothing the rotation, as it would "jump" from 180 to -180, and secondly it still wouldn't work if I re-center while in use. For example the output data would reach 180 first, and get stuck there, until the input data reaches that value as well (attached picture).
[image]
[image]

Is there a workaround? I hope I was clear, if there are any questions I can explain more.

Thank you for your time!

sthalik, 2018-11-04:
I could add a way to "center" by subtracting angles. It's in no way correct, but should work better than what we have right now. This should work for hatire, freepie and some HMDs.

The filter problems are secondary and the priority issue should be fixed first.

alex-k8, 2018-11-04:
> I could add a way to "center" by subtracting angles. It's in no way correct, but should work better than what we have right now. This should work for hatire, freepie and some HMDs.
> 
> The filter problems are secondary and the priority issue should be fixed first.

Would be great.

alex-k8, 2018-11-13:
Alright, I actually found a temporary fix. I added a button to my arduino, which when pressed resets the values that are sent to opentrack. That way both input and output are reset, and everything works as expected :smile:
Example code:
(button is on pinMode(2, INPUT_PULLUP))
if(digitalRead(2) == LOW){
    center_x=angle_x;
    center_y=angle_y;
    center_z=angle_z;
  }

  hat.gyro[0]=angle_y-center_y;
  hat.gyro[1]=angle_x-center_x;
  hat.gyro[2]=angle_z-center_z;

sthalik, 2019-01-03:
@132ikl I already implemented it.

sthalik, 2019-01-03:
Check the "experimental builds" wiki page for details.

132ikl, 2019-01-04:
[That page](https://github.com/opentrack/opentrack/wiki/experimental-builds) only contains a link to a dropbox that was last updated in November. What exactly are you referring to? Should we just build from source on unstable branch?

sthalik, 2019-01-04:
It should be working now. My Dropbox wasn't syncing.

132ikl, 2019-01-04:
Alright thanks, I'll try it out tomorrow.

closed 2020-01-27 by sthalik