opentrack/opentrack #848: [Suggestion]Use TrackIR output as Opentrack Input

Maetharin, 2018-11-24:
Hi there, I´ve been using TrackIR for the last few months, and I´m really happy with the consistency it provides me when compared to my earlier EDTracker/Aruco combo.

The one thing I really miss from Opentrack was the neck displacement feature, which TrackIRs TrueView is completely missing.

Would it be possible to use TrackIR output as an Input option in Opentrack? This way I could use Trueview with your neck displacement, by putting TrackIR acceleration of the xyz axes on 1:1 and increasing it again in opentrack.

sthalik, 2018-11-24:
These two data streams would conflict when using the freetrack protocol.

closed 2018-11-24 by sthalik

Maetharin, 2018-11-24:
still thx for answering