opentrack/opentrack #852: mouse emulation issue

uborapnik, 2018-12-24:
Sorry, I'm not sure if solution exist for my problem but I tried nearly everything...

I would like my mouse cursor to stay in place where I left it in-between toggling.
As it is now, if toggle OFF and move head, the mouse will jump and start moving from the point where my head is when I toggle ON.

I hope I explained issue clearly and you know what I mean. Am I missing something, is this possible or ?

sthalik, 2018-12-24:
Hey,

This makes sense. I'm introducing an option to disable the center behavior.

uborapnik, 2018-12-25:
Thats really good to hear! I hope it's not too hard to implement and happy holidays!

sthalik, 2019-01-03:
Bind the same button the centering and toggling. It should be enough.

uborapnik, 2019-01-04:
Tried that. Does not work.

sthalik, 2019-01-07:
I think centering with mouse output shouldn't result in any movement. Do you agree?

sthalik, 2019-01-07:
What if you used "center" instead of "toggle" completely? The center button functions while held and only stops working when it's depressed.

sthalik, 2019-01-11:
I need the game title where this is happening.

uborapnik, 2019-01-13:
It's already like this in windows. I wanna use it with gta:san andreas (multiplayer) though.

sthalik, 2019-01-18:
See the legacy mouse control mode in the latest snapshot at http://opentrack.misaki.pl. It's badly drifting though.

uborapnik, 2019-01-22:
Still the same issue, I recorded a video to give you better idea: https://youtu.be/oMphloijbWE

Let me know if there's anything else I can do to help.

sthalik, 2019-04-25:
Is your ingame resolution at least 1920x1080? There's a bad interaction between low resolution and mouse speed.

uborapnik, 2019-04-25:
2560x1440, same as windows

sthalik, 2019-04-30:
Please see if centering works in today's build. There was a bug with the cursor going out of bounds vertically.

closed 2020-01-27 by sthalik