opentrack/opentrack #853: Suggestion: Track by color!

Shadoware, 2018-12-30:
I dont use github frequently so Im novice here.

The software pointtracker works good but is generally hard to get a good hardware set to track infra red with quality, without interference and other camera/filter problems.

I would be a lot easier if its possible to use a clip with normal red led's for example and track it just by color like some kind of chromakey effect.

Is there a way to do it?

sthalik, 2018-12-30:
Yes, it's right there in PT tracker options.

closed 2018-12-30 by sthalik

MathijsG, 2019-01-20:
@Shadoware 

See:

[image]

Did it work out for you that way?