opentrack/opentrack #854: A test of beta1

Attigliuzzo, 2018-12-31:
I recorded a video ([this](https://youtu.be/jXCs2qd8-uI)) with Flightgear on my Debian machine to take an overview of the beta1. I setted calibration curves, moved the cross on the upper led and inverted movements for pitch and yaw. If i rotate on left or left-down there aren't issues, but if i try to look down or on the right the cross jumps as you can see in the video, anyway this version works well.
Thanks and Happy Christmas eve

sthalik, 2019-01-03:
Is it any better now in the unstable branch? That was one evil typo.

sthalik, 2019-01-03:
Thanks for additional info and closing your previous issue prematurely. See the wiki page named "experimental builds". It should be fixed.

Attigliuzzo, 2019-01-04:
@sthalik 
Yesss, your fix did the trick!!!!!! 
I did a double test:
1) (Debian+FlightGear+source code rev. 115): finally everything works perfectly, no more jumping cross.
2) (Win10+Falcon BMS+your experimental build 03/01 rev.110): there is only one issue, if i look back (180°) the view gives problem and i can see in that position the cross jumps. Probably because this build hasn't your fix. Anyway it works very well, like the stable version.

closed 2019-01-04 by Attigliuzzo

Attigliuzzo, 2019-01-05:
Ops!!! my fault, the "dynamic pose" option was disabled. Now it works perfectly, thanks