opentrack/opentrack #856: A link to a place for conversations about the software

garengllc, 2019-01-04:
Just found this software while trying to do some research on head tracking for 3-axis servos project I was thinking of doing.  I was trying to look around for somewhere where people discuss the opentrack software and projects related to it, but can't seem to find anything.

If there is such a place to bounce ideas around, a link to it would be helpful on one of the wiki pages.

sthalik, 2019-01-05:
You can look around forums for simulation titles. You won't find programmers or people deeply familiar with hardware though. It's less than once a year that another person contributes code for opentrack.

If you have people involved in your project already, feel free to use this issue tracker. It's treated more as a forum than as a bug tracker.

garengllc, 2019-01-07:
I wasn't very clear, sorry.  I signed in with my work account and not my personal, so I gave the wrong impression.  This is just a fun hobby project, so I am the only one working on it.  I also didn't mean developing the project in particular, so much as using the software as part of other projects.

I wanted to investigate trying to incorporate opentrack into taking movements off a phone and passing it onto something else for controlling servos.  I was thinking this might be the perfect avenue for recording motions for a 3-axis skull for Halloween.

sthalik, 2019-01-07:
You can, in fact, reuse opentrack's modules and there's an API for it. See `api/plugin-api.hpp` for just that.

There's an unfortunate dependency on Qt's GUI parts since each module includes a dialog for its configuration. You can avoid these bits by not instantiating any dialogs, but the shared library dependency will remain.

Other than that, ask rather than asking to ask. Cheers.

toozie21, 2019-01-07:
I was actually thinking more of running it as intended, and then piping the output somewhere else (via UDP).  I want to see what the data packets look like before I get to the next step, but I figure I am going to need to go to something intermediate before going directly to an Arduino, something like a Python script to handle the UDP and clean things up some.  This would be for real-time testing of the movements and potentially recording while doing the movements.

That plan may not work, if not, I will probably just record via python, and then handle the Arduino side of things after a complete recording is made.

Is there a way to set the interval of updates?  Or is it dependent on who is creating the original values (in this case a phone)?

Thanks for the help!

sthalik, 2019-01-07:
The output data is just `double stuff[6]` where individual elements are X, Y, Z, yaw, pitch, and roll consecutively. This is the entire datagram.

You can use this one-liner on a Unix machine or MSYS2/Cygwin: `nc -u -l -p 4242 | od -t x8`. Works for me on WSL. Of course Python with the `struct` module can do more with the datagrams. You can expect 250 packets a second regardless of the input device, with relatively little jitter.

Iknowlongnamerightwhatwasithinking, 2019-01-09:
Hello! 

Hey I am really pushing the boundaries of what opentrack is for. 

I have set up two cameras. and run two instances of OT and send headtracking data 
via freetrack protocol AND "stick tracking" data via Vjoy. 

So what the flipping heck? 

I have a joystick that moves a single LED around ( use the single pointtracker  mod I found ) 
and outputs stick data via Vjoy.  
It's BRILLIANT. 

The catch. 

The opentrack config files are in "my documents" ( I NEVER use my docs for anything btw) 
and so both instances of opentrack get mixed up by reading the same config files etc .
Ugh it's a pain in the rear. 

So I tried the "portable" version of opentrack and by golly did it look in "my documents" to read the config files... I would have thought "portable" would have meant it reads and saves it own deafault.ini 
in a folder in the "portable" exe's own folder. if you catch my drift. 

Anyway I am bashing this out because I really want this to work. It's almost working! 
( yes freetrack and vjoy are both going to my game ( RoF / BoS ) and stuff is working. ) 

Essentially if I could have TWO instances of open track running with non shared .ini files. 
( I can run them but they argue ....) 

Thanks! I love you all. No really. 

Robert Alexander.

sthalik, 2019-01-09:
> I have set up two cameras. and run two instances of OT and send headtracking data via freetrack protocol AND "stick tracking" data via Vjoy.

This is fine as well, just keep in mind that two instances of the freetrack protocol won't work along. Overall it's not right to launch the same protocol multiple times.

> The opentrack config files are in "my documents" ( I NEVER use my docs for anything btw)
and so both instances of opentrack get mixed up by reading the same config files etc .

You can send me a fix. Otherwise see below.

> So I tried the "portable" version of opentrack and by golly did it look in "my documents" to read the config files...

As per `options/global.cpp`, `bool is_portable_installation()` only fires if you create a file named `portable.txt` in the executable's directory. The "portable" archive is just an extracted installer, confusingly enough.

> Essentially if I could have TWO instances of open track running with non shared .ini files.

Definitely yes.

Iknowlongnamerightwhatwasithinking, 2019-01-09:
Hey thank you. 
I have two folders in root dir, first opentrack 01 and second opentrack 2. 
Can run both and write new .ini to respective dirs and looks ok. 
This of course due to portable.txt . 

OT one is outputting "freetrack" and OT two is outputting "vjoy" using the single point mod. 

I have notices that OT crashes almost every time on "start" when outputttin Vjoy. ???
Have to use other outputs till it settles down. 

Also unexpected quit randomly. Also using vjoy output. 

But I am very happy to get this far. : - ) 

This last thing: 

Very confused by the point tracker input model cap dimensions. and how it relates to the 
yellow cross on the tracking window. Changing values makes no sense. ( Blind person in room ...) 
the "start calibration" widget seems to make no sense in respect to where the cross ends up . 
Is their documentation to explain what is going on? 

I am a Freetrack refugee! I loved that program so much, but the "freetrack" dance to make it work wore me down.  I migrated to OT, just to have it start with no issues! I am happy now. 

Thank you for keeping the flame alive! Where do I donate? OT is a life saver! 

Thank you one million times. 

Salute! 

Robert Alexander.

sthalik, 2019-01-09:
> I have notices that OT crashes almost every time on "start" when outputttin Vjoy. ???
Have to use other outputs till it settles down.

What input do you use? It could be the input that's crashing. 

> Where do I donate? OT is a life saver!

Donate to the EFF. They deserve the money.

> Can I post a pic here of my project

Sure, why not.

> Yes even the pink octopus... though i did used to like the freetrack skull... it was edgy. : - )

The octopus was a stupid idea I came up with at 3AM or so. The facetracknoir logo had to be changed to *something* and I've had no good ideas. I don't like how the octopus is completely flat but no one else complained yet.

Iknowlongnamerightwhatwasithinking, 2019-01-09:
Hey. 

I use the point tracker input. Only crashes when output is set to Vjoy. 

I have come to be friends with the silly octopus. I even let it fly my Nieuport N11... 
Consider this a tiny complaint about how non 3d "Oatey the Octopus" is... very small even. 

S! 

R.A. 

PS. EFF.org?  I am picking it's that one. : - )

Iknowlongnamerightwhatwasithinking, 2019-01-09:
[image]

Still crashes + quits unexpectedly but, it is working. 
Had to roll back to previous version to accommodate single point mod, 
which is not portable. But it's ok. I can use the portable one for the second camera. 

Thank you! 

R.A.

Iknowlongnamerightwhatwasithinking, 2019-01-09:
Hello. 

Hey there is some new issue, not sure whether to post here of actually start a new thread. 

The Yellow cross thingy is different  across the 2.3.9. and 2.3.11
( 2.3.9 supports singlepoint mod and 2.3.11 is portable, so ican have two OT running ) 

[image]

Essentially 2.3.11 has mangled the way point tracker works? 
Both instances of OT are running the exact same default.ini file.  ( copied ) 
and yet the yellow cross fails to make any rational sense in 2.3.11 

Sorry to be a pain! 

R.A.

Iknowlongnamerightwhatwasithinking, 2019-01-09:
have installed 2.3.9 and 2.3.10 and 2.311 concurrently. 
2.3.9 uses my docs ini folder and both 2.3.10 and 2.3.11 both use their own ini folders.

2.3.10 can create own ini folder , this is great, 2 separate instances and separate .ini files.  
I fooled OT into to letting me install two different versions on same machine. It was fun. : -) 

By copy the same default.ini from 2.3.9 to 2.3.10 and 2.3.11 I find that 2.3.11 behaves quite different from the other 2. all three have exact same text in their respective .ini files. ( the yellow cross is way out ) 
So why is 2.3.11 different? 
R.A.

Iknowlongnamerightwhatwasithinking, 2019-01-10:
Sorry for so many comments. I am excited by success. 

Here are some pictures of my flight stick ( WW1 period style ) and the camera set up. 
I took the camera board out and mounted a magnet on the back, using sticky foam, and this magnet holds it onto the flight 4mm steel stick base and  I can adjust to zero it perfectly. Also I use the 640 for Y and 380 for X as pitch resolution is higher and stick back is important in Rise of flight. I decided to use Opentrack and the Camera to move away from hall effect sensors and MCU firmware with built in deadzone. 
So far it is very good and very very cheap. 
The camera and CL eye driver being most money spent, under $20 NZD for a stick that is very very accurate. : - ) 

Enjoy the pics! 

[image]
[image]
[image]
[image]
[image]

sthalik, 2019-01-10:
> The Yellow cross thingy is different across the 2.3.9. and 2.3.11 ( 2.3.9 supports singlepoint mod and 2.3.11 is portable, so ican have two OT running )
> Essentially 2.3.11 has mangled the way point tracker works?

Please provide some context.

> By copy the same default.ini from 2.3.9 to 2.3.10 and 2.3.11 I find that 2.3.11 behaves quite different from the other 2. all three have exact same text in their respective .ini files. ( the yellow cross is way out )
> So why is 2.3.11 different?

It's a bug with caps. I have it fixed already, for now use the third model tab and input point coordinates manually.

If you have two trackers, the second tracker should have its own settings group. Then you can use the "fusion" tracker to use position and rotation from each tracker separately.

Iknowlongnamerightwhatwasithinking, 2019-01-11:
Hey Stanislaw. 

Would be really good if there was some documentation I could read on how the pointracker works, with regard to, "model dimensions" and "model position". I am guessing the yellow cross is the point where the users head rotates on top of neck? Roughly below the ear. and there is some datum point which ties that to the LED frame ? It seems that this arcane knowledge has been lost in history.... can we get some actual description of what is going on here?  i.e Model position XYZ is informational a dead duck as it does not explain how the XYZ relates to the model. I think in free track it was the top led. Is this still valid? 

Thanks! 

R.A.

Iknowlongnamerightwhatwasithinking, 2019-01-11:
I could write up a doc if I had access to relevant information. 
That is no problem, I just would not want to get something wrong.
Working out how it works by inference is hard going. 
I may have a friend who might be interested in helping out a little bit, will talk to him. He lives code. 

Opentrack is a the last bastion of hope for those who wish to avoid paying big money for Track IR.  

Seriously I would pay a reasonable amount for what I have right now, of course. I use it everyday.  : - )

If you have links to information or anything... can you PM me? a forum??? Can I email you? 

Thank you so much for all your work !!! 

R.A.

sthalik, 2019-01-12:
> I could write up a doc if I had access to relevant information.

Model position aka "calibration" is a point in 3D space from where the rotation originates, aka [rotation axis](https://en.wikipedia.org/wiki/Rotation_around_a_fixed_axis). That is, there's the smallest amount of rotation when this point is selected as the rotation axis.

Now, this isn't the proper way of documenting it. Users totally turn off when confronted with something that vaguely resembles math from their school days. I understand it, my math education was a total trainwreck as well! But you can see that even a dummy like me can understand it in the lack of the psychological "math" block. The original code was written by Patrick back in facetracknoir days.

The purpose of calibration is to prevent rogue X and Y changes while yawing and pitching.

There are some caveats when using the calibration procedure:

- only yaw and pitch
- don't roll on purpose
- don't move on purpose
- take each pose (yaw left up, yaw left down, yaw right up, ...), not just turn your head and click "done"
- the calibration procedure draws statistical data from just 50-ish samples. it's not infallible
- if the model doesn't track normally, calibration is useless

Any project docs need to use relatively simple terms (not polysyllabic diarrhea like mine above), simplify things, but not stray *too* far from the truth. I don't expect the user to know what the axis of rotation is, I just want them to know what it's for and how to use it without bungling it up.

I have my own procedure and it's drunk stupid:

- start centered
- pitch down
- yaw left, right
- pitch normal
- yaw left, right
- pitch up
- yaw left, right
- yaw left
- pitch up, down
- yaw right
- pitch up, down
- Done! Only takes about 15 seconds.

> If you have links to information or anything... can you PM me? a forum??? Can I email you?

Feel free to email me (see user profile for email) if you intent to work on documenting the project. I don't have a problem with providing help via email in general but it's harder to manage than github, and sometimes messages get lost.

You can also search for "opentrack tutorial" on youtube and see the vast amount of information available. Also search for e.g. "opentrack calibration" and see another bunch of stuff.

This video is completely accurate: https://www.youtube.com/watch?v=ZDz-bholoMo

> Seriously I would pay a reasonable amount for what I have right now, of course. I use it everyday. : - )

I'm happy it works for you. I always say to others, donate to the EFF if you want to donate so bad. The project doesn't accept donations. I don't want to create a false expectation where donating would confer additional priority or privileges. It never would and never will, but some people assume that.

> Opentrack is a the last bastion of hope for those who wish to avoid paying big money for Track IR.

I'd have some things to say here about the latter, but let's remain civil and rant-free 😉. You can find some pages documenting what exactly happened back in 2008 or 2009.

Iknowlongnamerightwhatwasithinking, 2019-01-12:
Salute! 
Cool. I will do research and then write up notes etc. 
I am amateur mech eng and general eng/design/fab type person. 
( Applied ingenuity = engineering ) 
Will make a diagram with simple and easy to understand info. 

I never use calibration. 
It just seems to to do something non intuitive. ( i.e why is the cross there now???) 

So I totally believe the user needs to understand what is going on and then act accordingly. 
( become an engineer in your own right...or power user...) 

So... I will do some work. I like this kind of thing. 

In the end I could make a very small .jpg that could explain a few things. "explain to me how this works again.jpg" and a text file with additional secret useful information/procedure. 

I am interested in language and, long words versus short words and large character set versus small character set. It's all very interesting.... 

Will email you ASAP. 

Thanks! 

R.A.

sthalik, 2019-01-12:
> Applied ingenuity = engineering

People forget these days that "engineer" comes from "ingenuous" as a word in itself. Cheers 🍺!

> I never use calibration.
> It just seems to to do something non intuitive. ( i.e why is the cross there now???)

Imagine a helium-filled rubber balloon attached to a cord. You can twirl the cord to yaw it. The center of rotation is the place it's solidly attached to. Now imagine that balloon is your head, and the cord is the spine that lets you yaw and pitch. That's why this distinct cross is positioned somewhere inside your head.

Or imagine that you have an iron gimbal and allows for yaw and pitch at least. The axis of rotation is the solid, unmoving piece the gimbal is attached to (like a window has a gimbal letting it open sideways and upward (helpful with cats)).

> I am interested in language and, long words versus short words and large character set versus small 
> character set. It's all very interesting....

It comes down to style and framing when documentation is concerned.

Iknowlongnamerightwhatwasithinking, 2019-01-12:
Heh. I had to explain to an acquaintance who is a JS geek that anyone can be an engineer.
and some of the best engineers are self taught... : - 0 

So the cross is the centre point of the gimbal and the head is moving part ( in yaw and pitch ...) 
AND the head tracking points are displaced from that point.  ok I am doing a doodle now... 
I just have to draw this.... my eng ocd is kicking in. 

BRB.

closed 2020-01-27 by sthalik