opentrack/opentrack #857: UHD not working

toozie21, 2019-01-06:
I was trying to play around with the data and had the option setup to transmit out UDP, but I am not seeing anything.  I've setup a UDP client on my machine as well as tried sniffing with wireshark, but I don't see any data coming out.
I've set the transmit IP as myself, another machine on my network, and 127.0.0.1, no differences.

Maybe there is a configuration issue somewhere on my end, but I don't see how I could be messing this up.  I am running Windows 10 and checked to make sure that opentrack is allowed on my firewall list..

sthalik, 2019-01-07:
On Windows you're never able to sniff loopback traffic. You can only sniff traffic that touches any of the network interfaces. You can listen for UDP data as usual though. This works for me after allowing `netcat` through the Windows firewall.

```
/home/sthalik % nc -u -l -p 4242 | od -t x4
```

You can reproduce with Windows Subsystem for Linux or MSYS2/Cygwin. It's probably overkill though.

In any case I find it pretty dubious for the UDP protocol to fail to transmit data.

toozie21, 2019-01-07:
OK, I'll give that a shot this evening. I was questioning if Wireshark would catch things, and that was why I went with the UDP receiver route, but when that didn't work either, I was confused. I'll report back this evening. Thanks.

toozie21, 2019-01-08:
Well, no dice. I am going to blame windows.... I installed numerous UDP receive apps, made sure they were all allowed in the firewall and made sure the ports were open, but nope.

I need to throw another machine into the mix to see if that helps any, but at the end of the day, ai would like to do everything within this laptop, so it would just be a stopgap measure.

More attempts tomorrow.

toozie21, 2019-01-09:
So it looks like a python script to look for UDP packets works just fine.  I am not getting any negative values, but that is probably a script issue (Python isn't my strongest area).

But I think it is probably safe to say that UDP is working fine.

sthalik, 2019-01-09:
Thanks for the confirmation.

closed 2019-01-09 by sthalik