opentrack/opentrack #858: X and Y Axis bleed (only test with Aruco)

KalAbaddon, 2019-01-11:
2.3.10 Using the Aruco tracker the Y and X have almost exactly 50%/50% shared movement.  

i.e. if I try to move only on the Y axis the octopus will move on the X and Y balanced.  

and if I try to move on the X axis only it also registeres a balanced movement on the X and Y Axis.  

Once I downgraded I could perfectly move on the axis so it was not just picking up on my micro movements.

Everything else tracks fine. 
 
I have the paper marker in the same orientation as the guy in this link. 
https://github.com/opentrack/opentrack/issues/517
My setting are -22.5 and I have to reverse the Y axis.

For testing I downgraded by uninstalling 2.3.10 but keeping my profile.
I installed 2.3.1.4 and  it is INSTANTLY tracking straight now,  the x and y and completely independent.  and I still have it at -22.5 and reversed Y axis.

Here is my profile. Unfortunately I got this after downgrading so I do not know if it changed but all my setting seemed intact.
https://pastebin.com/Lzt2xHMm

Camera is~ 3 feet 6 inch front of me and ~ 6 inch up from center of aruco tracker.

sthalik, 2019-01-11:
Can you try 2.3.11-BETA2? There was a major change to how centering works. You can either have more or less axis bleed.

KalAbaddon, 2019-01-11:
In latest beta it does not even have the  0/+-22.5 deg options for the Aruco marker ( this was a really important feature for negating the deadzone created by the marker when dead on to the camera,  why was it removed?) for the Aruco so I can not correctly test.

sthalik, 2019-01-12:
> this was a really important feature for negating the deadzone created by the marker when dead on to the camera, why was it removed?

It didn't work for almost all users. Use regular Aruco instructions.

KalAbaddon, 2019-01-13:
Well poop...

I held my Aruco by hand so it is square and  it does zoom correctly so I guess that is fixed,  but it still has that issue around the identity pose.  And the 22.5 fixed it completely and it all works in 2.3.1.4.  

I will stick with 2.3.1.4 then for now.  I am disappointed the 22.5 option was removed.  I think it should be added back with maybe some more tlc to the Aruco setup page to help people on a budget since it was the best cheap setup I found by far.  But it is a moot point since 2.3.1.4 does what I want,  and I ordered a replacement setup 2 weeks ago that I dont need the Aruco for anymore.

sthalik, 2019-01-13:
> I ordered a replacement setup 2 weeks ago that I dont need the Aruco for anymore.

This is a good idea. Aruco data is of much lower quality than the PT tracker, for instance.

closed 2019-01-13 by sthalik