opentrack/opentrack #861: How to build a tracker?

Slion, 2019-01-29:
I'm about to start implementing a Kinect V2 tracker. I got all the input I need from the Kinect. I just need to implement the tracker itself. How can I build a tracker? Does it need Qt at all if I don't want a UI? Are trackers DLLs? Once build do I have to put it in a specific place? Can I just use an OpenTrack release and add my tracker to it?

sthalik, 2019-01-30:
> Does it need Qt at all if I don't want a UI? I've located that tracker-test code, I just want to be able to compile it, possibly without compiling the whole OpenTrack

You link against Qt one way or another given opentrack's dependencies. You have to make your own build, and there aren't any `.lib` files provided with the binary release either. Qt is the only hard dependency.

I recommend that you use the existing build system. Also, contributing ensures that your tracker stays working in case of any API/ABI changes, due to being maintained as part of the software.

sthalik, 2019-01-30:
> Say opentrack 2.3.11 BETA2, was it build with cygwin or VS, and does it matter?
> Will it be able to load a tracker I built using VS for instance?

It's a waste of time given ABI changes across Qt and MSVC CRT versions. Any Cygwin is never used, and mingw-w64 is not ABI-compatible with MSVC. Build the software normally, either mingw-w64 or MSVC works. In the former case you have working Qt builds provided already. GNU has no ABI fuckups like MSVC with several CRT versions, each distinct.

sthalik, 2019-01-30:
As per your response in https://github.com/opentrack/opentrack/issues/674#issuecomment-458591190, I'm concerned with regard to copyright matters. The original tracker has no license attached to it. According to the international treaties, all rights are reserved by default in such circumstances.

You may be able to use the libraries that tracker uses, but please don't base your code around that Java implementation, best not to read it at all in order to avoid accidentally creating a derivative work.

Furthermore, no GNU GPL code may be included. The GNU Lesser GPL license may be used, but in this case the library must be linked dynamically. These are legal matters and there can be no exceptions, sorry.

You can join the team without any hassle. For now, please send any working code you have as a pull request. We don't do branches much.

Slion, 2019-01-30:
I don't like that Java implementation. I'll go for plain old C++ and the only dependency will be on Microsoft Kinect V2 SDK.

Slion, 2019-01-30:
Was opentrack 2.3.11 BETA2 built with mingw-w64?

sthalik, 2019-01-30:
That's great! I recommend you trying out the new C++17 stuff, it's way more succinct and pleasant than ye olde C++98.

Just a few more boring legal things. I wish I never had to know these!

It's perfectly legal and appropriate to link against third-party SDKs when necessary to support the device.

All contributed code must contain a copyright statement. It's not necessary (though encouraged) to provide a real name. You can use a nickname if you want to. I recommend using the same [ISC license ](https://en.wikipedia.org/wiki/ISC_license#License_terms) as is used by most people contributing. It's not necessary to included that all-caps warranty statement.

> Was opentrack 2.3.11 BETA2 built with mingw-w64?

No, I use MSVC 2017 and it's fully C++17 compliant. You can build using mingw-w64 on your end. If anything fails, it's a bug in the build system and I need to fix it. Both toolchains are supported on Windows.

In the past I was easily able to cross from Linux and Windows for mingw-w64. MSYS2 is good for that.

Slion, 2019-01-30:
> Build the software normally
You mean as explained there:
https://github.com/opentrack/opentrack/wiki/Building-under-MS-Visual-C---2017-and-later

>  I use MSVC 2017
Awesome, that should make it easier for me then as I'm rather clueless when it comes to mingw.

Slion, 2019-01-30:
I just tried generating the VS2017 solution using CMake UI but it needs Qt5_DIR first.
You are saying I need to build Qt myself right, as explained on the wiki page except with Qt 5.12 now?

Slion, 2019-01-30:
Looks like Qt is pain to build too. They are saying it needs ActivePerl and Python. I hope it will work with Strawberry Perl...

sthalik, 2019-01-30:
Install official Qt for 32-bit Visual Studio 2015. Don't switch to `-MD` when building opentrack with cmake.

The path to `Qt5_DIR` is `lib/cmake/Qt5`.

Slion, 2019-01-30:
Haha, so I don't need to build Qt after all?
You distribution is 32-bit?

Slion, 2019-01-30:
> You're distribution is 32-bit?

Looks like it.

> Install official Qt for 32-bit Visual Studio 2015

I've installed Qt 5.12.0 using the online installer but it did not offer 32-bit for VS2015. It has it for VS2017 however. It does come with both libs and dlls.

Slion, 2019-01-30:
After installing OpenStack 2.3.11 beta 2, I replaced all Qt DLLs with the ones from the Qt SDK 32-bit VS2017 and OpenStack runs and works. Which means they are binary compatible and I should be able to just link against those.

sthalik, 2019-01-31:
I'd recommend using newest Visual Studio.

> You're distribution is 32-bit?

No reason not to. SSE3 is the minimum.

Slion, 2019-01-31:
> No reason not to. SSE3 is the minimum.

Fair enough, just needed to know. Though I reckon if you doing image recognition with OpenCV you would possibly get a measurable performance improvement running 64-bit.

Slion, 2019-01-31:
Just did a release build and I can run it just fine. Not just what's wrong with the debug build.

Slion, 2019-01-31:
All in all building the basics was really easy:

- Install Qt SDK (VS2017 32-bit)
- Configure CMake gui for VS2017
- Set Qt5_DIR to `C:/Dev/Qt/5.12.0/msvc2017/lib/cmake`, depending on your Qt installation folder.
- Generate VS2017 solution using CMake gui
- Do release build and install

I'm not sure why the wiki page needs to be so scary, it's rather off putting.
Now I'm guessing quite a few things are missing since I did not provide OpenCV directory and others but I like how you made those optional thus enabling a smooth minimal build.

Slion, 2019-01-31:
My release tracker test DLL is also binary compatible with your 2.3.11 beta 2 distribution even though it is only 33KB compared to 98KB.

Slion, 2019-01-31:
I pushed a first draft there:
https://github.com/Slion/opentrack/commit/ca6288adf8951af5bdaeeab41131ff560d4a0d4e
Though it's pretty useless as it is since I used the low definition API. I'll need to switch to the HD APIs.

Slion, 2019-01-31:
I pushed a first version using HD API. While it works the quality is far from what you get with TrackIR. It also rather unusable in low light.
I wonder if I would get better results feeding the buffers (colour, IR, depth) to OpenCV.

Slion, 2019-01-31:
I could also implement a version with IR points tracking. Maybe you could get me started with that too.

sthalik, 2019-02-01:
I don't think you need PCL. From the data you have in `c_FaceFrameFeatures`, it's easy to get approximate 3D coordinates and go for `cv::solvePnP`.

> I'm not sure why the wiki page needs to be so scary, it's rather off putting.

Once upon a time Qt didn't come with right ABI for mingw-w64 and latest Visual Studio. Feel free to edit the wiki page.

And kudos for figuring this on your own :)

> My release tracker test DLL is also binary compatible with your 2.3.11 beta 2 distribution even though 
> it is only 33KB compared to 98KB.

It's normal. Your libraries use `-MD`, mine use `-MT`. This works as long as they're dynamically linked.

From my experience, using `-MD` and including the redistributable is way larger than using `-MT` alone. The official builds prune unreferenced code from the CRT as well. This isn't possible with the dynamic CRT.

Slion, 2019-02-01:
> And kudos for figuring this on your own :)

You helped a lot.

> I don't think you need PCL. From the data you have in `c_FaceFrameFeatures`, it's easy to get approximate 3D coordinates and go for `cv::solvePnP`.

Since I switched to Kinect V2 Face HD API `c_FaceFrameFeatures` is not being used anymore. Kinect V2 has two similar APIs for face tracking, basic and HD. The basic one does not give you proper head position in space and is in general less accurate than the HD one. The HD one seems to be providing better quality yaw, pitch and roll. It's also possible to get a mesh and do a 3D face scan with the HD API.

I'm fairly happy with the current implementation but to be honest I don't think I will be using it much as it is not so accurate compared to TrackIR. However I would like to contribute it to master to make it available for people to try it out. Maybe it just does not like my face. It could also be that we find ways to optimise it overtime.

Here are my plans by order of priority:

1. Work with you to contribute that first implementation I called `tracker-kinect-face`.
1. Modify PointTracker to have it work using Kinect IR buffer.
1. Create another tracker called `tracker-kinect-cv` to see if we could get better results by using various Kinect raw output (RGB, IR, point cloud) and doing the processing ourselves using OpenCV for instance.

Along the way I could also find time to implement nice-to-have features and improvements in `tracker-kinect-face` such as providing the video stream for instance.

sthalik, 2019-02-01:
> I'm fairly happy with the current implementation but to be honest I don't think I will be using it much as it is not so accurate compared to TrackIR.

I don't have the hardware. Can you try it with the Accela filter? How much deadzone do you need? In particular, the deadzone helps way more with noise than the smoothing slider.

> Modify PointTracker to have it work using Kinect IR buffer.

Once we're at that stage, let me work on camera API support. In the end you're gonna be able to add a camera source from a tracker for use in another tracker.

I need to know if the IR data is single-channel or three-channel though.

> Along the way I could also find time to implement nice-to-have features and improvements in tracker-kinect-face such as providing the video stream for instance.

It can help you a lot to implement it while debugging the original code. You can visualize whatever data Kinect provides. Just ask if you have any problems. You can use `cv/video-widget` for this. It requires a 3-channel buffer but you can just copy the buffer thrice.

sthalik, 2019-02-01:
As for code review, I got you covered.

Slion, 2019-02-01:
>  In particular, the deadzone helps way more with noise than the smoothing slider.

Thanks for the tip. I just found the Accela filter settings. Maxing out both sliders improves usability a lot, that and mapping adjustments. I'm just getting started with OpenTrack itself so it could be I missed other obvious things.

> In the end you're gonna be able to add a camera source from a tracker for use in another tracker.

That would be awesome that would allow me to plug into PointTracker without touching its code.

> I need to know if the IR data is single-channel or three-channel though.

I can't tell for sure but visualising the IR stream gives you a black and white picture so I guess single channel.

Slion, 2019-02-01:
There seem to be quite a lag though when using Kinect Face Tracker in Elite Dangerous. Hard to tell where it's coming from without doing some serious profiling. Can I somehow adjust the polling rate of the tracker?

Slion, 2019-02-01:
Why is the pink billboard not pitching down? It only pitches up for both positive and negative values.

Slion, 2019-02-01:
How do suggest we deal with `tracker-kinect-face` binary dependencies?
It needs `Kinect20.Face.dll` and a folder called `NuiDatabase` containing about 50MB worth of data. They both need to be at the same level as `opentrack.exe`. They are from Microsoft's Kinect SDK.
Should we check them in our repository? They would also need to be referenced for installation by CMake and the installer itself I guess. Any tip on how to that would be helpful.

Slion, 2019-02-01:
Which version of OpenCV should I use?

sthalik, 2019-02-01:
> There seem to be quite a lag though when using Kinect Face Tracker in Elite Dangerous. Hard to tell where it's coming from without doing some serious profiling. Can I somehow adjust the polling rate of the tracker?

You can't. You're expected to do blocking operations (such as reading from a camera) from another thread, such that data is always available when `ITracker::data` is called.

The polling rate is 250 Hz (to prevent deconvergence for IMU trackers, and help filters) and it's not supposed to go down. You can see `logic/pipeline.cpp` for some bits on how the rate is maintained.

> How do suggest we deal with tracker-kinect-face binary dependencies?
> It needs Kinect20.Face.dll and a folder called NuiDatabase containing about 50MB worth of data. They
> both need to be at the same [...]

The .dll should work if you put it into the `modules` subdirectory. If `NuiDatabase` must be put at the same level, we'll figure something out. In particular, can you link to the SDK documentation? In the worst case it can be at toplevel.

> They would also need to be referenced for installation by CMake and the installer itself I guess.

It's enough to reference by CMake since the installer packs up the install directory. See how it's done by other modules that need additional files, like the freetrack protocol, or the realsense tracker.

You're expected to add a CMake cache variable for providing a path to the SDK, rather than any environment variable. You can initialize the cache variable from an environment variable in case the latter is set by the SDK's installer. Otherwise, leave it blank and set it up just like `Qt5_DIR`.

> Which version of OpenCV should I use?

Some 4.x or git master. I recommend the latter, and using static libraries.

sthalik, 2019-02-01:
What do you need opencv for, anyway? If it's only for the camera/widget API, that's not a good reason and the API needs to be changed.

>  I just found the Accela filter settings. Maxing out both sliders improves usability a lot, that and mapping adjustments. I'm just getting started with OpenTrack itself so it could be I missed other obvious things.

It depends a lot on your mappings, not just the filter. If yaw and pitch have a small range, the noise is more visible as well. Compare yaw going to 180 at 10 degrees, and yaw going to 180 at 18-ish degrees. The latter is more stable.

Is it "bearable" or "well" with maxed out Accela settings? I understand this is a face tracker so by definition it's less accurate than PT or Aruco.

> Why is the pink billboard not pitching down? It only pitches up for both positive and negative values.

It's not perspective projection so moving straight up and straight down (with no yaw) looks the same.

> That would be awesome that would allow me to plug into PointTracker without touching its code.

It's just a bit of work on my end. I need to make absolutely sure of Kinect buffer's channel count, though. There are different things that need to be done in either case. opentrack traditionally used 3-channel buffers only, and camera-based trackers need adapting. Right now they unconditionally grayscale images, same for the display widget code.

sthalik, 2019-02-01:
That's great!

Are you working on the face tracker, or should we go for camera API directly?

Slion, 2019-02-01:
> Are you working on the face tracker, or should we go for camera API directly?

As mentioned above my priority should be on contributing a first version of `tracker-kinect-face`. I'll notably be looking into incorporating your CMake recommendations and create a push request for code review whenever I'm ready.

Though Camera API is probably not far behind there is really no rush. I reckon implementing video stream support for `tracker-kinect-face` would be a good exercise for me before looking into extending PointTracker.

Slion, 2019-02-01:
> The .dll should work if you put it into the `modules` subdirectory.

 Yes it works. Both DLL and database can live in the `modules` folder.

Slion, 2019-02-01:
> In particular, can you link to the SDK documentation?

[Kinect V2 SDK docs](https://docs.microsoft.com/en-us/previous-versions/windows/kinect/dn799271(v%3dieb.10)).

Slion, 2019-02-01:
> It's not perspective projection so moving straight up and straight down (with no yaw) looks the same.

It would be nice to fix this. That would be great for testing.

Slion, 2019-02-01:
Should I commit those binaries dependencies under `\opentrack\bin\kinect`?

sthalik, 2019-02-01:
> It would be nice to fix this. That would be useful for testing.

Generally it works if you add any yaw or roll. The pose-widget module isn't done too well and should be rewritten from scratch in a more efficient manner. I'm thinking of some `QTransform` stuff.

Fortunately, it only consumes CPU while the software is visible on screen.

> Should I commit those binaries dependencies under `\opentrack\bin\kinect`

Put them under `tracker-kinect/redist`. I'd also like know whether the Kinect SDK allows for redistributing these files. Please locate relevant licensing information and put it in `3rdparty-notices`. I can't do releases with stuff that's not legal to redistribute.

I recommend naming it `tracker-kinect` in the directory structure and similarly in tracker metadata.

> Though Camera API is probably not far behind there is really no rush. I reckon implementing video stream support for `tracker-kinect-face` would be a good exercise for me before looking into extending PointTracker.

I've done some work on making `cv_video_widget` accept single-channel images. I haven't done any work on the pt/aruco trackers yet.

Slion, 2019-02-02:
> I recommend naming it tracker-kinect in the directory structure and similarly in tracker metadata.

I called it `tracker-kinect-face` as looking ahead there could be several trackers using Kinect one way or another. This one is using the Kinect SDK face API. As mentioned I'm already contemplating a `tracker-kinect-ir` and possibly a more complex `tracker-kinect-cv`.

Slion, 2019-02-02:
> You're expected to add a CMake cache variable for providing a path to the SDK, rather than any environment variable. You can initialize the cache variable from an environment variable in case the latter is set by the SDK's installer. 

Just fixed that.

Slion, 2019-02-02:
I went for a quick test flight in ED and after using mapping to add some dead zone around the centre position it's  totally usable. That's my first impression anyway. Is it possible to provide users something like a default profile for that tracker mapping and Accela settings?

sthalik, 2019-02-02:
It's not a proper review though. I'm gonna do one once you're satisfied with the code and I have the SDK running.

Slion, 2019-02-02:
Video preview is working. I'll now have to clean up the code and hunt down memory leaks.

sthalik, 2019-02-03:
If you need to initialize stuff just once, use a static variable like that:

```
void initialize_my_stuff_() {
...
}
void initialize_my_stuff() {
    static int guard = initialize_my_stuff_(), 0;
    (void)guard;
}
```

opentrack uses this idiom a bit, and it's even thread-safe.

sthalik, 2019-02-03:
You're doing great! The commits look promising.

Slion, 2019-02-03:
Here we go:
https://github.com/opentrack/opentrack/pull/869

Slion, 2019-02-03:
> I need to know if the IR data is single-channel or three-channel though. 

I've had a quick look at the Kinect IR buffer and it's a single channel of 16-bit shorts. However the way forward is most certainly to convert it to RGBA before handling it over to Point Tracker. Unless Point Tracker can also handle single channel buffers but even then it would most likely need to be converted and copied anyway.

sthalik, 2019-02-16:
@Slion unless you add a copyright statement to `kinect_face_tracker.h`, or `.cpp` I won't be able to include your code in the next public release. Thanks.

sthalik, 2019-02-28:
@Slion should I include the redistributable files (`msvcr110.dll` and `msvcp110.dll`) or are they installed by the redistributable?

Slion, 2019-02-28:
I did not realise those were needed but I guess it makes sense that the Kinect DLLs have a dependency against those. On my developer machine they are installed in Windows/System32 most certainly the doing of VS2012. They are not part of the Kinect redist but they are VS2012 redist.
On my machine I can also find them in:
`C:\Program Files (x86)\Microsoft Visual Studio 11.0\VC\redist\x86\Microsoft.VC110.CRT`

Slion, 2019-02-28:
Ideally we should do a similar CMake setup than what we did for the Kinect SDK.
* Have a CMake config variable named  SDK_VS2012 pointing to VS2012 installation directory.
* There an env variable named VS110COMNTOOLS that's pointing to `C:\Program Files (x86)\Microsoft Visual Studio 11.0\Common7\Tools\`. We could use it to initialise SDK_VS2012.
* From there build the path to the redist is easy. 

Want me to go ahead and implement that before you release?

sthalik, 2019-02-28:
Please don't. The build system can't presuppose having VS 2012 anywhere, or even a Windows installation. I'd expect a working build when cross-compiling from Linux or FreeBSD. It's also legal to bundle the .dlls themselves with the program.

sthalik, 2019-02-28:
My original question was as to whether the redistributables are installed by installing the Kinect *redistributable* itself. In which case it's not necessary to bundle them with opentrack (they're already there).

Slion, 2019-02-28:
The whole Kinect stuff is Windows only. It won't build on any other platform and has been thus configured in CMake.

> My original question was as to whether the redistributables are installed by installing the Kinect _redistributable_ itself. In which case it's not necessary to bundle them with opentrack (they're already there).

No, those DLLs are VS2012 redist and are not provided by the Kinect SDK. They are provided with VS2012 or the [standalone corresponding redist package](https://www.microsoft.com/en-us/download/details.aspx?id=30679).

sthalik, 2019-03-01:
Then I think it's best they're included in the source tree like that. I'm greatly opposed to including redist installers due to how long they take to finish. It's the same reason why opentrack is built with the `-MT` switch.

Slion, 2019-03-03:
I usually avoid including binaries under source control. As mentioned above, to me the proper way should be similar to what we did for the Kinect SDK binaries. Then again I have no idea how your build machine is setup.

Slion, 2019-03-03:
I like to avoid putting binaries under source control. As mentioned above, to me the best solution should be similar to want we did for the Kinect SDK libraries. Just let me know if you need anything from me.
[VC-redist-11-0.zip](https://github.com/opentrack/opentrack/files/2922957/VC-redist-11-0.zip)

sthalik, 2019-03-03:
See if Kinect works for you in BETA3. We're about to make a release. Other than that, we're good.

Slion, 2019-03-03:
Just tested beta 3 and it's working great. I guess we should close that issue.

[20 more replies]

closed 2019-03-03 by Slion