opentrack/opentrack #862: Is OpenTrack working with Qt 5.12?

Slion, 2019-01-30:
OpenTrack wiki mentioned Qt 5.10. I was wondering if using the actual 5.12 version instead was doable?

sthalik, 2019-01-30:
Yes, the official and experimental builds now use 5.12.

sthalik, 2019-01-30:
Feel free to ask if you have any questions regarding building the software.

closed 2019-01-30 by sthalik

Slion, 2019-01-30:
That's great news. Looks like you could use some help maintaining the wiki pages among other things.
I did ask a bunch of other questions on a few other issues. Feel free to answer them :)
Looks like I'll get opportunities to contribute if only to the wiki as I'm trying to implement possibly multiple trackers for Kinect. Maybe it makes sense that I join the team so that I could just work on branches in that repository without having to clone it.