opentrack/opentrack #865: Keeps crashing

drpain, 2019-01-30:
When using PointTracker + Freetrack 2.0

```Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	opentrack.exe
  Application Version:	0.0.0.0
  Application Timestamp:	599200f1
  Fault Module Name:	opentrack-tracker-pt.dll
  Fault Module Version:	0.0.0.0
  Fault Module Timestamp:	59920101
  Exception Code:	c0000005
  Exception Offset:	000e3bae
  OS Version:	6.1.7601.2.1.0.256.1
  Locale ID:	7177
  Additional Information 1:	0a9e
  Additional Information 2:	0a9e372d3b4ad19135b953a78882e789
  Additional Information 3:	0a9e
  Additional Information 4:	0a9e372d3b4ad19135b953a78882e789

Read our privacy statement online:
  http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409

If the online privacy statement is not available, please read our privacy statement offline:
  C:\Windows\system32\en-US\erofflps.txt
```

sthalik, 2019-02-01:
Hey,

What camera are you using? Does this happen each time?

Unfortunately a bug like this requires more data. Please install Visual Studio 2017 Community. I'll provide further instructions then.

drpain, 2019-02-01:
Hi hi, it happens almost every time. I sometimes have to "Start" like 20 times before it actually starts successfully. 

What seems to help is 2 things, but I can't isolate it because it just crashes less often:
- Closing the TrackIR process
- Only enabling my cams streaming software after Open Track has started.

If that helps at all

sthalik, 2019-02-01:
> Only enabling my cams streaming software after Open Track has started.

What software is this?

> Closing the TrackIR process

This shouldn't matter. The process only keeps running because opentrack crashed, and wasn't able to close it.

sthalik, 2019-02-03:
I should add it to the FAQ. Some apps inject their own code into all processes and that crashes opentrack. The software isn't responsible for arbitrary code messing around with its private data.

sthalik, 2019-02-03:
Can you confirm whether the crashes go away if you uninstall DroidCam?

closed 2020-01-27 by sthalik