opentrack/opentrack #866: How to add Kinect V2 IR support to PointTracker

Slion, 2019-01-31:
I'm considering adding Kinect V2 IR support to PointTracker. I'm not sure though where to get started.
Basically I'll need to feed the Kinect V2 IR buffer to PointTracker somehow.

sthalik, 2019-02-01:
There's not yet an API for additional camera sources. I've been meaning to add this for some time. 

Are you ok with starting with tracking faces as per your initial implementation?

sthalik, 2019-02-01:
>> There's not yet an API for additional camera sources. I've been meaning to add this for some time.
> We could work on that together. If you can point me in the right direction in the code that would help a lot already.

Please don't work on that on your own. In any case, it's gonna be possible to expose additional camera sources from any module the same way `options/metatype` does that, or the migration module.

The webcam source will be `camera.h` from the point tracker module. The rest will be exposed by other modules, using the same base API.

> I'll be trying to complete and contribute that tracker based on Kinect v2 Face API. I'll need your help and review to make sure it's ready for master. It's working already albeit not as well as I hoped. I'm pretty sure I could get better face tracking using third party libraries to process the Kinect inputs.

Should we continue in #861 or here?

closed 2019-02-16 by sthalik