opentrack/opentrack #867: Can't load Qt DLLs when building Debug target

Slion, 2019-02-01:
When building Debug target against pre-built Qt binaries as explained [there](https://github.com/opentrack/opentrack/issues/861#issuecomment-459213842) OpenTrack won't start complaining about missing Qt DLLs.

Both Release and RelWithDebInfo targets work as intended. This is an issue for both 32-bit and 64-bit variants.

Slion, 2019-02-01:
Looks like that fixed it for Qt but it complains about a bunch of other "missing" DLLs:
* opentrack-logic.dll
* opentrack-migration.dll
* opentrack-pose-widget.dll
* opentrack-user-interface.dll

sthalik, 2019-02-01:
They should be in your install directory.

Slion, 2019-02-02:
> They should be in your install directory.

They are, but like the Qt libs before your fix they are not being picked up for whatever reason.

sthalik, 2019-02-02:
Try with Dependency Walker.

Slion, 2019-02-07:
I got the debug build running. What it needs is the *d.dll files for Qt and OpenCV. I probably missed that 'd' suffix until now. So I guess we should somehow fix our CMake install so that it picks up the proper DLLs. Also it would be nice to support install for multiple targets something like `/install/$(TargetName)` instead of installing all target into that one install folder.

Slion, 2019-02-07:
Debug allows inspecting variables which mostly did not work RelWithDebInfo.

Slion, 2019-02-07:
Sadly when starting PoinTracker in Debug with my Kinect camera it crashes but that's another topic.

sthalik, 2019-02-09:
There's no particular functionality for the `.sln` generator as Visual Studio now has native CMake support. Worse yet, switching the install prefix would regen your entire solution.

I have almost no problems using the debugger with optimized builds. See `cmake/msvc.cmake` for my toolchain file. YMMV.

It should now copy over the right libraries, see `cmake/opentrack-qt.cmake`, function `otr_install_qt_libs`

```cmake
get_property(path TARGET "${i}" PROPERTY "LOCATION_${CMAKE_BUILD_TYPE}")
```

Slion, 2019-02-09:
> Visual Studio now has native CMake support
I had no idea, got to try it out.

Slion, 2019-02-09:
I had a quick try at the CMake feature from Visual Studio. It's looking great but it seems to have it's own set of problems. For some reason it did not build the projects depending on OpenCV even though I provided the directory in that json file together with Qt. Moreover, I'm used to the .sln workflows and layout, things are quite different when loading a folder. Also while CMake gui is self explanatory, showing you your cache variables and options the VS approach isn't and you basically need a guide or dig deep in the CMake files to work out what you need to configure. It's also adding an extra layer of complexity in the form of those json files.

Slion, 2019-02-09:
> For some reason it did not build the projects depending on OpenCV

My mistake, they were built. It was just missing the OpenCV DLLs since you are usually linking statically against OpenCV. It did copy the proper d.DLL for Qt. Though that's still not working for the generated solution.

sthalik, 2019-02-10:
I agree it can be a pain learning CMake. In other IDEs it's way easier and there's no need for any manually-edited `.json` files.

The `INSTALL` target in VS should copy debug .dll's for Qt. Can you give some steps to reproduce?

There's no logic for copying over OpenCV libraries. I'm not sure whether it's practical to use the shared libraries given their size and lost potential for link-time optimization.

sthalik, 2019-02-10:
In general you can leave the CMake stuff to me. I made some fixes to your tracker's build system.

The video widget code needn't depend on opencv. It's just written that way given separate overloads for `QImage` and `cv::Mat`. I'm gonna fix it so there's no need to install opencv for Kinect support.

Please add a copyright statement in your header, thanks. This is important, lack of any copyright statement reserves all rights. Thanks.

Slion, 2019-02-10:
> Can you give some steps to reproduce?

https://github.com/opentrack/opentrack/wiki/Developers'-Quick-Start-Guide-for-Windows

Slion, 2019-02-10:
> I'm gonna fix it so there's no need to install opencv for Kinect support.
I did not dare asking but you read my mind.

Slion, 2019-02-10:
> Please add a copyright statement in your header, thanks. This is important, lack of any copyright statement reserves all rights. Thanks.

Just headers or sources too? Shall I just copy the one from other files in the project?

Slion, 2019-02-10:
> There's no logic for copying over OpenCV libraries. I'm not sure whether it's practical to use the shared libraries given their size and lost potential for link-time optimization.

Not having to build OpenCV yourself is a great advantage, faster build time, easier setup.

sthalik, 2019-02-11:
> Just headers or sources too? Shall I just copy the one from other files in the project?

I recommend you use the short license from other files, but you must replace the author. Keep the format with the authorship's year. You needn't provide a legal name (a nickname will do) or an email address.

sthalik, 2019-02-11:
> Not having to build OpenCV yourself is a great advantage, faster build time, easier setup.
Maybe we could add a CMake option to switch between static and dynamic OpenCV linkage. If you really want to preserve static linkage that is.

I should be able to detect library type and act accordingly.

sthalik, 2019-04-05:
@Slion shared opencv build should work ok for you now.

sthalik, 2019-04-05:
Reopen if there are any issues. Thanks!

closed 2019-04-05 by sthalik