opentrack/opentrack #870: [Tracking issue?] Roll not following yaw/ roll turns into pitch looking to the side/ roll inverted when looking back

Goteer, 2019-02-03:
I've been having this problem since last month. Before then, it worked perfectly fine, and there was no change in hardware when it started.

When tracking, roll seems to _always rotate the head as if yaw were to equal zero_. At first I thought it was a visual glitch in the preview, but after testing in steamvr (what I was using opentrack for previously) it was made clear to me that the preview was being correctly displayed:

-When looking forward, roll functions normally

[image]

-When looking to the side, roll functions as pitch (Even though the real pitch value doesn't change)

[image]

(Using pitch instead of roll)

[image]

-When looking back, roll is inverted

[image]

Recentering simply changes what direction "forward" is, but the issue persists.

I was using my phone with either freepieIMU or udp receiver for headtracking, but while troubleshooting I've had the same problem with the aruco marker, so I can only guess that the phone's sensors aren't the problem.
[image]
(Added yaw for illustration purposes)

Imgur album with all the images (Plus freepieIMU images showcasing the same issue)
https://imgur.com/a/fhRtN9I

**Things I've tried:**
-Enabling/disabling relative translation
-Using diferent axis for roll (using pitch input for roll for testing, issue still present)
-Using different versions of opentrack: 2.3.11 beta2, 2.3.10, 2.3.9, 2.3.0, multiple 2.3 prereleases (just in case)
-Using both relative and absolute centering in older releases (absolute centering appears to fix the problem, but introduces axis bleeding instead)
**-Using a different machine (got opentrack on my laptop and had the same problem, somehow. Tested freepieIMU and aruco. Both machines have windows 10, up to date. Still got to test on windows 7 and possibly on linux)**

**Current settings:**
[image]

I wanted to test the newest unstable but I have no clue how to build it (got mingw-w64 and cmake, but I get errors. Probably me doing something dumb)

Slion, 2019-02-06:
Try those instructions if building on Windows:
https://github.com/opentrack/opentrack/wiki/Developers'-Quick-Start-Guide-for-Windows

That pink octopus does some funny things at times. Notably, it does not pitch down, always up.
You may want to try it in your application.

Goteer, 2019-02-06:
> 
> 
> Try those instructions if building on Windows:
> https://github.com/opentrack/opentrack/wiki/Developers'-Quick-Start-Guide-for-Windows
> 
> That pink octopus does some funny things at times. Notably, it does not pitch down, always up.
> You may want to try it in your application.

Just tried the latest experimental release since it was uploaded the same day as the latest commit. After fiddling for a while with the settings just in case, I'm still having the same issue. Currently installing what I need to build it myself just in case.
Thanks for replying!

Goteer, 2019-02-09:
I've just tried bypassing any and all external programs entirely and using a udp driver for steamvr. Having the same issue, which leads me to believe opentrack is doing exactly what it's supposed to.
I think steamvr just updated how sensor data is processed and the drivers for diy stuff just haven't updated.
I'm still mystified by the octopus acting like it is. I also know my phone's sensors aren't the problem, since I've tested with other phones and with aruco too.

sthalik, 2019-02-09:
I think it's the rotation order that makes a difference. From your screenshots, it looks valid if yaw is applied prior to roll. Have you checked if it's any different in the latest experimental builds? See the wiki.

The code that checks for rotations outside the `+-180` range may be buggy. It should wrap around nicely but who knows. It reminds me, I should rewrite it.

Goteer, 2019-02-12:
> I think it's the rotation order that makes a difference. From your screenshots, it looks valid if yaw is applied prior to roll. Have you checked if it's any different in the latest experimental [...]

Just tried the 02/11/2019 experimental, having the exact same issue
https://imgur.com/a/ejQME8K

Also, yaw rotation seems to jump from 70 degrees to -180 (going from 0 to 180), but functions correctly when going the other way (from 0 to -180), but that might just be my phone having weird sensor values (Though freepie shows symmetrical values).

sthalik, 2019-02-13:
This is gimbal lock. Is roll of sixty realistic?

Goteer, 2019-02-15:
@sthalik I'm not sure what you mean by realistic, all I can tell you is that the roll issue is permanently there, from 0 to 90-something (above that the phone's roll sensor is useless so I wouldn't know), same with negative values. Roll value stays under that +-90(give or take), following the real roll accordingly.

I just quickly searched for what gimbal lock is, I guess pitch and roll are the parallel axis in this case, but the numbers shown appear correctly.

sthalik, 2019-02-16:
> @sthalik I'm not sure what you mean by realistic, all I can tell you is that the roll issue is permanently there

Your second screenshot has a yaw value of `-260`.

> I just quickly searched for what gimbal lock is, I guess pitch and roll are the parallel axis in this case, but the numbers shown appear correctly.

Not exactly, but you can just avoid excessive roll values. You won't be able to achieve that high a value anyway when using the tracker (unless lying on the floor or something).

Goteer, 2019-02-16:
@sthalik

>Your second screenshot has a yaw value of `-260`.
I can't replicate that yaw value. I'm trying with 2.3.0, 2.3.10 and the February 11th experimental, but neither raw or game data goes above 180 or below -180, always wraps around. Roll issue remains.

The roll value itself seems to function correctly all the way up to +-80º regardless of yaw or pitch (as in follows my phone's angle correctly) , even when the octopus shows otherwise.  The roll issue is still there , and remains effective regardless of how much roll there is: As little as 5 roll still becomes "5 fake-pitch" (without affecting pitch values). 
https://imgur.com/4DRHfpJ (raw yaw is pitch and raw pitch is yaw)

Pitch and yaw are also following my phone's angles correctly (with the exception of the yaw jump mentioned two messages ago). 
In 2.3.10 there's some axis bleeding from pitch to roll, but it's completely gone in the 2.3.11 beta.

I find it strange that the numbers show the correct angles, but the octopus and the output to the steamvr driver both apparently apply roll in the wrong order.

sthalik, 2019-02-16:
> I find it strange that the numbers show the correct angles, but the octopus and the output to the steamvr driver both apparently apply roll in the wrong order.

There's no single correct order. Each order is correct, in its own way.

Goteer, 2019-02-20:
@sthalik Is there a way to manually change the order? Or is it just hardcoded and too much work to have as an option?

sthalik, 2019-02-20:
@Goteer, it's much work in distinct places. Your sender app should convert to the right order prior to sending.

Goteer, 2019-02-20:
@sthalik Ah, ok then. Thank you for pointing me in the right direction. I'll see what I can do about it, since at this point it wouldn't be an opentrack issue. 
Thank you again.

closed 2019-02-20 by Goteer