opentrack/opentrack #871: Hotkeys stop working after program has been running for some time

HerraTohtori, 2019-02-05:
Hello,

I've been using FreeTrack until now. I built a new PC and migrated to Windows 10, and of course FreeTrack doesn't work properly on that platform any more (no matter what I try, I can only get 30 FPS out of my PS3 Eye on FreeTrack - everything else runs normally).

So, I managed to set up the tracking profile on opentrack to produce acceptable results. However, after using the program for a while, I noticed that the centering hotkey didn't work any more. Further investigation revealed that other hotkeys didn't work either. This applies to both in-game and when window focus is on opentrack.

Basically, immediately after I start the program and tracking, everything works correctly for some time - until it doesn't. The time it takes for hotkeys to stop working seems to be random, however I have noticed that trying to do basically anything in opentrack (options, mapping, input/output/filter settings) while tracking is running will very likely instantly cause hotkeys to stop working. But even when left undisturbed, the hotkeys will stop working at some point.

Snce the centering function especially is regularly used, this seems like something that other people would have noticed and reported if it was something that affected everybody, but I can't for the life of me figure out what could possibly cause it on my system.

I've tried the 2.3.10 release as well as 2.3.11 beta version, and both exhibit the same issue. I could start digging up older versions to figure out whether they can work, but they might introduce other issues that are already fixed in the newer builds.

I've tried using keyboard commands with and without modifier keys, as well as a joystick button. All work initially, but obviously it's not reliable. So the input method itself doesn't seem to be the problem. It feels more like the opentrack program for some reason "stops listening" for hotkey inputs.

What would be the best way to find out what's causing this? Are there any debugging builds available for download, or would I need to compile my own or something?

Len62, 2019-02-05:
Are you running OT at the administrative level?

HerraTohtori, 2019-02-05:
I am now, thanks for the suggestion. It didn't occur to me because the program itself was "working" and not throwing any visible errors about requiring to be run as admin.

It seems to have made at least some difference - going through the menus no longer insta-wrecks hotkeys. I'll stress-test this solution more - if it works through several hours of gameplay, then I would say it's no longer an issue that threatens the actual use of the program.

Is this a known issue? I mean, the readme didn't say anything about it - is this just one of those cases where regular users know the program needs to be run as admin?

Len62, 2019-02-05:
This was the first time I heard of this issue. Anytime you are running multiple programs it's a good idea to run them at the admin level so they have system wide access. Prevents you from having to worry about what program has the "focus".

HerraTohtori, 2019-02-05:
I took a "nap" and woke up about 7 hours later, and the hotkeys are still working.

This is kind of an interesting issue. I mean I don't have any problem running a program with admin access, but this kind of thing definitely should work without it. As far as I understand, running multiple programs on admin level shouldn't technically be any different than running multiple programs on user level, if they all have the same priority. I would say it is in fact very bad practice to run programs as admin without a very good reason for why it is required. Both from user (risk of compromising the system increases if the first attempted solution to any issue is "run as admin", and from development perspective it's just sloppy even though Microsoft is making some things abnormally inconvenient). Running a program in admin mode should only be required if the program is doing something that needs elevated permissions.

Program having "focus" shouldn't prevent things like hotkeys from working. Other programs with hotkeys (Discord, Teamspeak, etc.) generally don't have this issue, so clearly there's some problem with opentrack itself that causes the loss of hotkey inputs in regular user mode.

Most importantly, though, a program (or a feature in it) should definitely not just silently fail with no errors. There are probably some errors generated somewhere, but without debug build a regular user will never see them. The only indication the user has is that the hotkeys stop working, seemingly randomly, but with shocking regularity.

Well, at least this issue is documented now, and there is an interim fix that not only appears to work pretty well but will also probably give coders an idea on what to fix so that the hotkeys keep working when run as regular user.

sthalik, 2019-02-09:
@HerraTohtori,

The software doesn't stop listening for hotkeys on its own. In the event keyboard or joystick can't be polled, they're attempted to be reacquired. As the first thing you can try launching with debug messages shown, i.e. drag the executable onto a Windows console window. The messages will appear while the program runs.

> What would be the best way to find out what's causing this? Are there any debugging builds available for download, or would I need to compile my own or something?

Install Visual Studio 2017 with C++ support. Once you have it, I'm going to guide you through the debugging process. The debug info is already included in the app.

HerraTohtori, 2019-02-09:
> @HerraTohtori,
>
> The software doesn't stop listening for hotkeys on its own. In the event keyboard or joystick can't be polled, they're attempted to be reacquired. As the first thing you can try [...]

Okay. Running the program directly from console does not pop up any debug messages relating to polling failure of any input devices. Hotkeys still fail when opentrack is running without admin privileges.

From a naíve standpoint (not knowing much about what's involved) this suggest that opentrack thinks the polling is working normally, but for some reason the hotkey inputs are either not reaching the program (intercepted by something?) or the program receives the inputs but then doesn't do the correct thing with them (i.e. ignores them).

Considering there are no debug messages at all, and that everything seems to work normally with admin privileges, I'm inclined towards something in the Windows 10 system on my computer is doing something that's causing this issue by preventing opentrack from "seeing" the hotkey inputs.

Since I'm not running any third party antivirus programs at the moment, the most obvious suspects were Windows Defender and/or UAC settings.

So I went into Windows Defender settings and under "Virus & threat protection settings" I disabled "Real-time protection", while opentrack was still running with hotkeys not working. Immediately, hotkeys resumed working normally.

After turning real-time protection back on in Windows Defender settings, and messing around a bit in opentrack, the program soon froze and I had to end the process manually.

My conclusion at this moment is that there's probably nothing wrong in opentrack itself (at least on this particular issue). Rather, it's probably a false positive by Windows Defender that's causing the issues when running without elevated privileges.

**Steps to reproduce:** Have a Windows 10 Home version 10.0.17763 installed with no third party firewalls or antivirus programs. UAC settings are set on "Never notify" (I like to live dangerously). Start opentrack, make sure there are some keys bound to hotkeys, and after some time the hotkey functionality should disappear.

**Solutions:** Run opentrack with admin privileges. (confirmed to work in my case)
Add opentrack directory (or possibly just the executable?) to Windows Defender's "Virus & threat protection" exceptions? (currently testing this)
Install third party antivirus that supercedes Windows Defender and doesn't identify opentrack as something that needs to be defended against? (will probably work)

> Install Visual Studio 2017 with C++ support. Once you have it, I'm going to guide you through the debugging process. The debug info is already included in the app.

If my above "logical debugging" is correct, I'm not sure if there's any info to be debugged. Console didn't show any messages during the time when hotkeys were not working, so if it's really Windows Defender "intercepting" the hotkeys, then the program probably just isn't seeing any inputs at the time.

At that point the "debugging" of the issue would turn into a question of "how to make Windows Defender not mess things up" and I really don't have enough programming experience to be of much help on this issue. Other than being able to reproduce the issue and test different builds.

sthalik, 2019-02-10:
@HerraTohtori,

Thanks for the excellent analysis. Let's leave the issue as open and see if more people have the same problem.

UAC-wise, I recommend turning it off through Group Policy since the minimum level still doesn't run all programs as the administrator user.

> At that point the "debugging" of the issue would turn into a question of "how to make Windows Defender not mess things up" and I really don't have enough programming experience to be of much help on this issue. Other than being able to reproduce the issue and test different builds.

Check today's build. There were some issues where an input device (keyboard and joysticks alike) weren't reacquired properly. It should now poll the devices as specified.

There are some core changes as part of a major rework. If there's a problem I can make a branch with only the Direct Input changes.

closed 2020-01-27 by sthalik