opentrack/opentrack #873: Issue with IL-2 BOX

gerriraz, 2019-02-16:
Hello, I'm experiencing some problem using opentrack 2.3.10 with IL-2 BOX sim and more precisely I've no yaw at all and a very little pitcH ( 5° /8° output) movement: at same time I've tiny x movement when rotating the head. Octopus move correctly on opentrack interface. I gave a try to every suggestion I've found on the net and on forum  but nothing worked. Hany help will be appreciated.
Best regards

Len62, 2019-02-16:
What are your current settings in OT? And have you made any adjustments in the mapping tab?

sthalik, 2019-02-16:
Is anyone with that game available?

Len62, 2019-02-16:
It's pretty much the only game I play. Never experienced the issue the OP has. I'll try and help him out, I do not have any issues in IL2 BOX.

gerriraz, 2019-02-16:
I had tried pretty much all mapping settings finding best settings for me (for roll,x,y,z all works well).
For yaw and pitch axis ( the most important for this game) nothing to do. Pls let me know what Ot settings are you interested in and I'll tell you. Strange thing is that the pink octopus is moving right as I expect. Trying with another tracking program I've same issue so I started thinking it could be a sim program communication problem.

Len62, 2019-02-16:
Are you running both Opentrack and IL2  at the admin level? As far as settings, what is your set-up? IR clip and PS3 camera (what i use), or some other configuration?

gerriraz, 2019-02-16:
I gave a try to another flight sim (X-Plane X) and all works perfectly...some minor adjustments but all works. So I'm more convinced it's a game configuration issue if other people with same sim and same tracking program have all working. Now I've just to discover what's wrong....

gerriraz, 2019-02-16:
Running both as admin and PS3 Camera with IR led clip

Len62, 2019-02-16:
Try this ini file to see if it helps. It's not a zip file, just change "zip" to "ini". Also, check the value for the key "old_trackir =" in the IL2 startup.cfg file, mine is 0.
[2.3.10.zip](https://github.com/opentrack/opentrack/files/2872277/2.3.10.zip)

sthalik, 2019-02-16:
Maybe it can be related to "Use NPClient, use freetrack", or `old_trackir` options?

sthalik, 2019-02-16:
> One other thing to try, If your output is Freetrack 2.0 enhanced, make sure "enable both" is selected in the output settings.

This is the compatibility setting, but it's not any game's correct setting.

gerriraz, 2019-02-17:
Solved....after a lot of time on the net searching for a solution I saw a post where someone said that unplugging the joystick opentrack worked as it should. So I started thinking about a conflict between opentrack and joystick command settings. ...and it was! More precisely I had assigned horizontal (turn head) and vertical ( bow head) commands to my tm twcs throttle hat and this setting overwelmed incoming inputs from opentrack. I just disabled that coupling and opentrack starts working as a dream. Thank you for your help....cheers

Len62, 2019-02-17:
Glad you got it sorted out!

sthalik, 2019-02-17:
@gerriraz glad you got it sorted out.
@Len62 thanks for helping out.

closed 2019-02-17 by sthalik