opentrack/opentrack #874: Kinect video preview is broken

Slion, 2019-02-16:
Possibly since the dependency on OpenCV was removed.

Slion, 2019-02-16:
The issue is with the code in `video_widget::paintEvent` which keeps deleting the bitmap that was provided replacing it with one filled with grey. I'm not sure what's the proper fix.

sthalik, 2019-02-16:
It's unfortunate that the preview isn't of the right size to begin with. With opencv it's `malloc`-free and efficient, but `QImage::scaled` allocates.

sthalik, 2019-02-16:
@Slion please add the copyright statement in the header. It can't go into a binary release until you do.

sthalik, 2019-02-16:
I removed this entire block since `texture` can be guaranteed consistent at this point.

closed 2019-02-16 by sthalik

Slion, 2019-02-17:
> It's unfortunate that the preview isn't of the right size to begin with. With opencv it's malloc-free and efficient, but QImage::scaled allocates.

I agree, idealy our render loop should be free of allocations too. Though I'm not sure why a larger texture could not be scaled in smaller one for rendering without allocation, surely Qt can do that one way or another.