opentrack/opentrack #876: How to set up Wiimote tracking?

David-Carranza, 2019-02-22:
I connected my Wiimote to my PC and when I try to start tracking all I see is this image?

[image]

What does this image mean?
Sorry if this isn't the right place to post this I just don't know where else to get help.

sthalik, 2019-02-22:
Are you able to pair the device through Bluetooth normally?

David-Carranza, 2019-02-22:
Yeah I added my device through the control panel and when I see my bluetooth devices this is what I see

[image]

sthalik, 2019-02-22:
@cpuwolf can you assist please?

Fakespeare78, 2019-03-09:
That is 100% identical to my problem. 

Windows 10 (64), tried 2 different Bluetooth dongles (old one + new one), can't get past that point. Downloaded Dolphin to test the Wiimote and it works.

sthalik, 2019-03-14:
Please see today's experimental build at <<http://opentrack.misaki.pl>>. It's labelled as `WII-TEST`.

Fakespeare78, 2019-03-14:
Wiimote inits fine, I can move it around and see a horizontal line moving, but as soon as I point it towards an IR source (tracking clip, IR remote control or turn on a lighter), Opentrack exits without error message. 

```
Protokollname: Application
Quelle:        Application Error
Datum:         14.03.2019 19:55:32
Ereignis-ID:   1000
Aufgabenkategorie:(100)
Ebene:         Fehler
Schlüsselwörter:Klassisch
Benutzer:      Nicht zutreffend
Computer:      i7-860-Reboot
Beschreibung:
Name der fehlerhaften Anwendung: opentrack.exe, Version: 0.0.0.0, Zeitstempel: 0x5c8a577e
Name des fehlerhaften Moduls: opentrack-tracker-wii.dll, Version: 0.0.0.0, Zeitstempel: 0x5c8a560f
Ausnahmecode: 0xc0000005
Fehleroffset: 0x0006f0c1
ID des fehlerhaften Prozesses: 0x1580
Startzeit der fehlerhaften Anwendung: 0x01d4da9773ec2b56
Pfad der fehlerhaften Anwendung: C:\Users\Mathias\Downloads\opentrack-20190314-TEST-WII\install\opentrack.exe
Pfad des fehlerhaften Moduls: C:\Users\Mathias\Downloads\opentrack-20190314-TEST-WII\install\modules\opentrack-tracker-wii.dll
Berichtskennung: 50d6b80c-d399-4488-be32-c74670b0236c
Vollständiger Name des fehlerhaften Pakets: 
Anwendungs-ID, die relativ zum fehlerhaften Paket ist: 
Ereignis-XML:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" />
    <EventID Qualifiers="0">1000</EventID>
    <Level>2</Level>
    <Task>100</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2019-03-14T18:55:32.962903200Z" />
    <EventRecordID>49542</EventRecordID>
    <Channel>Application</Channel>
    <Computer>i7-860-Reboot</Computer>
    <Security />
  </System>
  <EventData>
    <Data>opentrack.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>5c8a577e</Data>
    <Data>opentrack-tracker-wii.dll</Data>
    <Data>0.0.0.0</Data>
    <Data>5c8a560f</Data>
    <Data>c0000005</Data>
    <Data>0006f0c1</Data>
    <Data>1580</Data>
    <Data>01d4da9773ec2b56</Data>
    <Data>C:\Users\Mathias\Downloads\opentrack-20190314-TEST-WII\install\opentrack.exe</Data>
    <Data>C:\Users\Mathias\Downloads\opentrack-20190314-TEST-WII\install\modules\opentrack-tracker-wii.dll</Data>
    <Data>50d6b80c-d399-4488-be32-c74670b0236c</Data>
    <Data>
    </Data>
    <Data>
    </Data>
  </EventData>
</Event>
```

```
Protokollname: Application
Quelle:        Windows Error Reporting
Datum:         14.03.2019 19:55:35
Ereignis-ID:   1001
Aufgabenkategorie:Keine
Ebene:         Informationen
Schlüsselwörter:Klassisch
Benutzer:      Nicht zutreffend
Computer:      i7-860-Reboot
Beschreibung:
Fehlerbucket 2237560293950948829, Typ 1
Ereignisname: APPCRASH
Antwort: Nicht verfügbar
CAB-Datei-ID: 0

Problemsignatur:
P1: opentrack.exe
P2: 0.0.0.0
P3: 5c8a577e
P4: opentrack-tracker-wii.dll
P5: 0.0.0.0
P6: 5c8a560f
P7: c0000005
P8: 0006f0c1
P9: 
P10: 

Angefügte Dateien:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER8685.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER8771.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER8781.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER877F.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER879F.tmp.txt

Diese Dateien befinden sich möglicherweise hier:
C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_opentrack.exe_b0e11e824b515d155c9a473085dab23d8e21de9f_f8f8192f_32778e55

Analysesymbol: 
Es wird erneut nach einer Lösung gesucht: 0
Berichts-ID: 50d6b80c-d399-4488-be32-c74670b0236c
Berichtstatus: 268435456
Bucket mit Hash: ad67623e169ac0a06f0d693b747705dd
CAB-Datei-Guid: 0
Ereignis-XML:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Windows Error Reporting" />
    <EventID Qualifiers="0">1001</EventID>
    <Level>4</Level>
    <Task>0</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2019-03-14T18:55:35.045879600Z" />
    <EventRecordID>49543</EventRecordID>
    <Channel>Application</Channel>
    <Computer>i7-860-Reboot</Computer>
    <Security />
  </System>
  <EventData>
    <Data>2237560293950948829</Data>
    <Data>1</Data>
    <Data>APPCRASH</Data>
    <Data>Nicht verfügbar</Data>
    <Data>0</Data>
    <Data>opentrack.exe</Data>
    <Data>0.0.0.0</Data>
    <Data>5c8a577e</Data>
    <Data>opentrack-tracker-wii.dll</Data>
    <Data>0.0.0.0</Data>
    <Data>5c8a560f</Data>
    <Data>c0000005</Data>
    <Data>0006f0c1</Data>
    <Data>
    </Data>
    <Data>
    </Data>
    <Data>
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER8685.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER8771.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER8781.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER877F.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER879F.tmp.txt</Data>
    <Data>C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_opentrack.exe_b0e11e824b515d155c9a473085dab23d8e21de9f_f8f8192f_32778e55</Data>
    <Data>
    </Data>
    <Data>0</Data>
    <Data>50d6b80c-d399-4488-be32-c74670b0236c</Data>
    <Data>268435456</Data>
    <Data>ad67623e169ac0a06f0d693b747705dd</Data>
    <Data>0</Data>
  </EventData>
</Event>
```

sthalik, 2019-03-14:
@Fakespeare78 please use Task Manager after it crashes, go to the Details tab and select "Create dump file" on `opentrack.exe`. Then send it here, thanks!

sthalik, 2019-03-14:
The file from your crash event is probably exactly that, a minidump: `C:\ProgramData\Microsoft\Windows\WER\Temp\WER8685.tmp.dmp` from the text pasted.

Fakespeare78, 2019-03-14:
Sorry @sthalik - might have been false alarm, I was not able to reproduce the problem; tracking is working fine, no more crashes. 

The .dmps are not there, I can offer a "Report.wer", but deleting and pairing the Wiimote again - or rebooting? - seems to have solved my problem. 

Thank you very much for your help, time and all the work!

sthalik, 2019-03-14:
@Fakespeare78 thanks for the information. If it crashes ever again, please send the dump. In particular, maybe it crashes when the device isn't paired to begin with, or is in some particular state.

rogamaral, 2019-03-26:
Same problem, same image as reported by David-Carranza.
Windows 7 64
Opentrack 2.3.10
Bluetooth dongle Asus USB-BT400

Is it possible to be a problem related to the device brand/type?
Thanks.

sthalik, 2019-03-28:
@rogamaral can you see if the problem persists in the newest experimental build? It's going to become the next release eventually. See <<http://opentrack.misaki.pl>>.

cpuwolf, 2019-04-10:
opentrack-20190401_01.7z it crashes while points are there.  if no points, it runs fine

sthalik, 2019-04-10:
@cpuwolf can you provide a minidump please?

cpuwolf, 2019-04-11:
FAULTING_IP: 
opentrack_tracker_wii!GetDialog+43ab1
0033:58e092c1 <Effective machine and debuggee state conflict, disassembly not possible>

EXCEPTION_RECORD:  ffffffffffffffff -- (.exr 0xffffffffffffffff)
ExceptionAddress: 0000000058e092c1 (opentrack_tracker_wii!GetDialog+0x0000000000043ab1)
   ExceptionCode: c0000005 (Access violation)
  ExceptionFlags: 00000000
NumberParameters: 2
   Parameter[0]: 0000000000000001
   Parameter[1]: 0000000000000000
Attempt to write to address 0000000000000000

PROCESS_NAME:  opentrack.exe

ADDITIONAL_DEBUG_TEXT:  
Use '!findthebuild' command to search for the target build information.
If the build information is available, run '!findthebuild -s ; .reload' to set symbol path and load symbols.

FAULTING_MODULE: 0000000076c30000 kernel32

DEBUG_FLR_IMAGE_TIMESTAMP:  5c9c8b02

ERROR_CODE: (NTSTATUS) 0xc0000005 - 0x%08lx

EXCEPTION_CODE: (NTSTATUS) 0xc0000005 - 0x%08lx

EXCEPTION_PARAMETER1:  0000000000000001

EXCEPTION_PARAMETER2:  0000000000000000

WRITE_ADDRESS:  0000000000000000 

FOLLOWUP_IP: 
opentrack_tracker_wii!GetDialog+43ab1
0033:58e092c1 <Effective machine and debuggee state conflict, disassembly not possible>

MOD_LIST: <ANALYSIS/>

LAST_CONTROL_TRANSFER:  from 0000000000000000 to 000000007498cb49

FAULTING_THREAD:  ffffffffffffffff

BUGCHECK_STR:  APPLICATION_FAULT_NULL_POINTER_WRITE_WRONG_SYMBOLS_ZEROED_STACK

PRIMARY_PROBLEM_CLASS:  NULL_POINTER_WRITE

DEFAULT_BUCKET_ID:  NULL_POINTER_WRITE

STACK_TEXT:  
03fede40 00000000 03fedc9c 00000000 7659346e wow64!Wow64LdrpInitialize+0x8d5

STACK_COMMAND:  kb

SYMBOL_STACK_INDEX:  0

SYMBOL_NAME:  opentrack_tracker_wii!GetDialog+43ab1

FOLLOWUP_NAME:  MachineOwner

MODULE_NAME: opentrack_tracker_wii

IMAGE_NAME:  opentrack-tracker-wii.dll

BUCKET_ID:  WRONG_SYMBOLS

FAILURE_BUCKET_ID:  NULL_POINTER_WRITE_c0000005_opentrack-tracker-wii.dll!GetDialog

WATSON_STAGEONE_URL:  http://watson.microsoft.com/StageOne/opentrack_exe/0_0_0_0/5ca2011b/opentrack-tracker-wii_dll/0_0_0_0/5c9c8b02/c0000005/000492c1.htm?Retriage=1

Followup: MachineOwner
---------

0:008:x86> lmvm opentrack_tracker_wii
start             end                 module name
58dc0000 58f7f000   opentrack_tracker_wii   (export symbols)       opentrack-tracker-wii.dll
    Loaded symbol image file: opentrack-tracker-wii.dll
    Image path: C:\Users\user\Downloads\opentrack-20190401_01\install\modules\opentrack-tracker-wii.dll
    Image name: opentrack-tracker-wii.dll
    Timestamp:        Thu Mar 28 16:51:14 2019 (5C9C8B02)
    CheckSum:         001A904A
    ImageSize:        001BF000
    File version:     0.0.0.0
    Product version:  0.0.0.0
    File flags:       0 (Mask 0)
    File OS:          0 Unknown Base
    File type:        0.0 Unknown
    File date:        00000000.00000000
    Translations:     0000.04b0 0000.04e4 0409.04b0 0409.04e4

sthalik, 2019-04-13:
Fortunately this was a benign assert hit in opencv. I'm compressing today's build and it should be up within half an hour.

cpuwolf, 2019-04-22:
opentrack-20190413_01.7z works fine

manfromjam1, 2019-12-15:
I also have the same picture when I try to use my wiimote.  I created a dump file and tried to zip it but it was still too large.  7zip files are not accepted.
windows 10 64 bit
Asus BT400 Bluetooth dongle
OpenTrack 2.3.11 beta3

manfromjam1, 2019-12-28:
> Same problem, same image as reported by David-Carranza.
> Windows 7 64
> Opentrack 2.3.10
> Bluetooth dongle Asus USB-BT400
> Is it possible to be a problem related to the device brand/model?
> Thanks.
> EDIT: After install de Asus software/driver it works.

I think my problem is the bt400.  I see this in event viewer for my HID compliant game controller
Device HID\{00001124-0000-1000-8000-00805f9b34fb}_VID&0002057e_PID&0330\8&2536cd3&2&0000 was not migrated due to partial or ambiguous match.

Last Device Instance Id: HID\VID_046D&PID_C513&MI_01&Col04\9&314197ff&0&0003
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Location Path: 
Migration Rank: 0xF000FC00FFFFF100
Present: false
Status: 0xC0000451

closed 2020-01-27 by sthalik