opentrack/opentrack #878: Opentrack software won't open

spicyborland, 2019-02-24:
I'm totally unable to start the opentrack software. I've tried installing/running it as admin, adding exceptions to my security, etc but when I double click on the software it does nothing, although the program does appear in my task manager. I'm on win 10 64bit. Any ideas?

sthalik, 2019-02-24:
Make sure you're not using GIGABYTE OC GURU or Moonlight software. They're known to interfere with all running processes, and affect opentrack in a wrong way.

I have more ways to diagnose this with other tools but let's try the most common issue first.

spicyborland, 2019-02-24:
hey there, just fixed it - was a conflict with an old version of some magnification software that another user had installed. Thanks though

closed 2019-02-24 by spicyborland