opentrack/opentrack #879: Opentrack stopped working

famellad, 2019-02-25:
OS: `Windows 10 Pro (v. 1809)`
Versions affected: `2.3.10, 2.3.11 beta 1, 2.3.11 beta 2`

I looked through the open and closed issues and couldn't find anything similar (and reasonably recent), if I missed some archaic old thread that happens to be this very problem, I apologise in advance :(

**Problem description**
Tonight I turned on my computer and Opentrack wasn't communicating with my games anymore, prior to restarting it worked flawlessly on both Euro Truck Simulator 2 and Elite: Dangerous, yet now it doesn't work on neither.

**How to reproduce**
I honestly have no idea how to reproduce this problem, the one thing I remember doing differently last time was saving my profile because I wanted to try different values on the filter. I was using 2.3.11 beta 1 when this problem first appeared, I have since tried other versions (including 2.3.10, which I also had working some months ago)

**Things I've tried**
- Rebooting
- Uninstalling and reinstalling
- Uninstalling, deleting references from registry and reinstalling
- Uninstalling, deleting references from registry, rebooting and reinstalling

**Additional info**
The tracker seems to be working fine, it seems to be just the output that's failing.
On 2.3.11 it says "Unknown Game" in the title bar at all times, even when a supported game is running.

Feel free to ask for any further information regarding my setup, right now I'm gonna go catch some sleep, and many thanks in advance!

Len62, 2019-02-27:
What are you using for your output settings?

famellad, 2019-03-01:
Freetrack 2.0 enhanced, with TrackIR hidden

Len62, 2019-03-01:
Try enabling both.

famellad, 2019-03-01:
Doesn't work 😢

famellad, 2019-03-01:
Okay I got it to work by downloading someone else's profile off the internet, but I can't, for the life of me, figure out what's different on this one

sthalik, 2019-03-01:
Send it as attachment here, please.

Len62, 2019-03-01:
@famellad Glad you got it sorted out. Strange issue, make sure you send the faulty ini as sthalik requested.

lukiluk100, 2019-08-01:
Sorry for digging up the topic, but i have the same problem. I see "Unknown Game" and nothing else.
Can anyone give me Opentrack profile?

cromsan, 2019-09-13:
Resuelto el anterior problema al desistalar e instalar, 
Nuevo problema: el Opentrack ya no se comunica  con mi juego Xplane 11.
He desistalado opentrack y vuelto a instalar la versión 2.3.10, pero  el problema continua. Uso el freetrack 2.0.
 No sé cómo mandarles el ini?

cromsan, 2019-09-13:
Les mando el log. Si necesitan algo más, pidanmelo.
[Log.txt](https://github.com/opentrack/opentrack/files/3609825/Log.txt)

sthalik, 2019-09-13:
That file's not an opentrack profile. Please use English exclusively.

cromsan, 2019-09-14:
Sorry; where is the opentrack profile so I can send.

Len62, 2019-09-14:
On Windows 10 it's in the documents/opentrack folder.

cromsan, 2019-09-14:
I send what I found in documents/ opentrack, on file type ZIP (It was a file type RAR)

[opentrack-2.3.zip](https://github.com/opentrack/opentrack/files/3613209/opentrack-2.3.zip)

cromsan, 2019-09-14:
De: Len62
Enviado: sábado, 14 de septiembre de 2019 22:31
Para: opentrack/opentrack
CC: cromsan; Comment
Asunto: Re: [opentrack/opentrack] Opentrack stopped working (#879)

On Windows 10 it's in the documents/opentrack folder.

Len62, 2019-09-16:
Does OT work ok with some games but not others, or does it just not start due to the startup error? Is OT not finding the camera? If it does not start at all try deleting all of the ini files except the default ini.

cromsan, 2019-09-17:
I only play xplane 11. Opentrack do not open now. I have put opentrack on disk D. ¿where is the ini files, I do not find them?

Enviado el: martes, 17 de septiembre de 2019 0:41
Para: opentrack/opentrack <opentrack@noreply.github.com>
CC: cromsan <c.romerosanzdemadrid@gmail.com>; Comment <comment@noreply.github.com>

Len62, 2019-09-17:
The ini files are located in the documents/opentrack folder. What is the exact error message you get when you run OT? What drive do you have Windows installed on?

cromsan, 2019-09-17:
In the opentrack, doc, I have: 3rdparty-notices
                                                contrib
                                                settings
                                                README.md
I do not get any error message, because now OT does not open.
I do not understand "what drive do you have Windows intalled". I have not installed any driver.

Len62, 2019-09-17:
They are not in opentrack/doc folder. See this pic: 
[image]

Next, are you using windows 10 operating system, or any other OS, what hard drive is it installed on?

cromsan, 2019-09-17:
[image]

cromsan, 2019-09-18:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

cromsan, 2019-09-18:
Sorry for making the same 11 times.

cromsan, 2019-09-18:
I have to put this files of opentrack 2.3.10 into the opentrack on disk D I send before?
[image]

Len62, 2019-09-18:
At this point I think it would be best to just delete everything you have installed for OT. Then I would make sure that your camera is working properly. Once you've done that I would re-install OT, but install it on your C: drive. Once you have done that see if OT will at least run without crashing. What type of tracking device are you using?

cromsan, 2019-09-18:
Thank you, Len62. I will do what you say when I want to come back to opentrack.  For the moment, i am going to rest of opentrack.

Len62, 2019-09-18:
Ok. You can probably close this thread then.

cromsan, 2019-09-19:
Yes, I close this thread.

closed 2020-01-27 by sthalik

pilpo, 2021-01-18:
> 
> 
> Once you've done that I would re-install OT, but install it on your C: drive (Disco local (C:)). 

That did the trick for me! Thanks!