opentrack/opentrack #880: crop video feature request

R5fan, 2019-02-25:
It would be useful if we could limit the area in the video feed where the tracking points can be extracted. 
Im using opentrack in a club flight simulator and its almost impossible to remove some light bleeding and reflections in the upper and side parts of the video feed that results in tracking errors, but that area is pretty far from where tracking LEDs are. If I could just tell opentrack to ignore these regions, that would solve the issue. I experimented with resolution and FoV settings for my PS3eye camera, but that doesnt seem to do anything.

Similarly, if not yet implemented,  it may help if we can pinpoint a center location where the LEDs should be expected, and give more weight to pixels detected in that region than to pixels on the extremities.

MathijsG, 2019-02-28:
With the daylight filter installed you still can have annoying reflections or unwanted lights, so I understand the usecase. FOV and resolution aren't going to crop the visible image.

sthalik, 2019-02-28:
Using the manual threshold mode should help you. As long as your min/max point size is restrictive enough, the light won't come through.

R5fan, 2019-03-31:
The camera already has its IR filter removed and believe me, I tried every possible treshold setting, but sunlight reflections are just too powerful and Im not able to position the camera in a way that always eliminates them at the edges. I may just have to tape the edges of the camera...

MathijsG, 2019-03-31:
Did you set up the camera with PS3 eye test tool to reduce the exposure and gain and set colors to white?

sthalik, 2019-03-31:
> Did you set up the camera with PS3 eye test tool to reduce the exposure and gain and set colors to 
white?

From my experience direct sunlight passes through stronger than the LEDs around 850 nm wavelength.

closed 2020-01-27 by sthalik