opentrack/opentrack #881: Receiving data via UDP in Max/Msp

RasmusEklund, 2019-02-26:
Hi,

I have been able to receive tracking data via UDP in Max/Msp from Opentrack. However, I am not sure how to interpret the data I am receiving. I.e. what does the yaw, pitch and roll and xyz represent for each message I am receiving? 

[image]

RasmusEklund, 2019-03-05:
I figured out after some indepth testing and research. I will upload a Max patch here that receive the UDP messages from Opentrack and converts them to xyz and pitch yaw and roll. In case others have the same problem and want a solution :) 
[OrientationExtractor.zip](https://github.com/opentrack/opentrack/files/2930657/OrientationExtractor.zip)

sthalik, 2019-03-05:
> I have been able to receive tracking data via UDP in Max/Msp from Opentrack. However, I am not sure how to interpret the data I am receiving.

These are yaw, pitch, roll, x, y, and z as double-precision floats. The floats are always transmitted little-endian so on embedded arches you may have to flip the bytes around. Rotation in degrees, position change in centimeters.

RasmusEklund, 2019-03-06:
I didn't flip the bytes around to begin with, so that's why it didn't make sense in the beginning. It works great now. Thanks!

closed 2019-03-07 by sthalik