opentrack/opentrack #882: Beta3: "Cannot load library opentrack-tracker-pt.so"

Attigliuzzo, 2019-03-01:
Finally i can test the beta3 on my debian testing. It compiles well but i cannot load the opentrack-tracker-pt.so library, launching opentrack from command line:

> [/media/attila/back_up/flightgear/opentrack/api/plugin-support.hpp:178]: library "pt" failed: "Cannot load library /usr/local/opentrack/bin/../libexec/opentrack//opentrack-tracker-pt.so: (/usr/local/opentrack/bin/../libexec/opentrack//opentrack-tracker-pt.so: undefined symbol: _ZN7pt_impl30qInitResources_tracker_pt_baseEv)"

Thanks

sthalik, 2019-03-01:
In the build directory within the directory structure you should have a file named `qrc_tracker_pt_base.cpp.obj`. Please run the `nm` tool like that:

```
nm qrc_tracker_pt_base.cpp.obj
```

and send the file as an attachment here, or to pastebin. Thanks.

sthalik, 2019-03-01:
Please see if this helps.

[pt-namespace-fix.diff.txt](https://github.com/opentrack/opentrack/files/2920316/pt-namespace-fix.diff.txt)

sthalik, 2019-03-01:
Please see if this diff helps.

[pt-namespace-fix2.diff.txt](https://github.com/opentrack/opentrack/files/2920327/pt-namespace-fix2.diff.txt)

sthalik, 2019-03-01:
I was able to reproduce and fix it. Thanks for the report!

commit b4217a4a (2019-03-01): tracker/pt: fix link error due to namespace

Reported by: @Attigliuzzo
Fixes: #882
closed 2019-03-01 by sthalik