opentrack/opentrack #883: Linker warning as error options::value

Slion, 2019-03-03:
Just pulled and getting the following error:

`1>------ Build started: Project: opentrack-options, Configuration: Debug x64 ------
1>   Creating library C:/Dev/GitHub/bin/opentrack-64/options/Debug/opentrack-options.lib and object C:/Dev/GitHub/bin/opentrack-64/options/Debug/opentrack-options.exp
1>tie.obj : warning LNK4217: locally defined symbol ??B?$value@N@options@@QEBANXZ (public: __cdecl options::value<double>::operator double(void)const ) imported in function "void __cdecl options::tie_setting(class options::value<double> &,class QDoubleSpinBox *)" (?tie_setting@options@@YAXAEAV?$value@N@1@PEAVQDoubleSpinBox@@@Z)
1>LINK : error LNK1218: warning treated as error; no output file generated
1>Done building project "opentrack-options.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 5 up-to-date, 0 skipped ==========`

Slion, 2019-03-03:
Actually there are a bunch of other similar warnings:

```1>------ Build started: Project: opentrack-options, Configuration: Debug x64 ------
1>   Creating library C:/Dev/GitHub/bin/opentrack-64/options/Debug/opentrack-options.lib and object C:/Dev/GitHub/bin/opentrack-64/options/Debug/opentrack-options.exp
1>tie.obj : warning LNK4217: locally defined symbol ??B?$value@N@options@@QEBANXZ (public: __cdecl options::value<double>::operator double(void)const ) imported in function "void __cdecl options::tie_setting(class options::value<double> &,class QDoubleSpinBox *)" (?tie_setting@options@@YAXAEAV?$value@N@1@PEAVQDoubleSpinBox@@@Z)
1>tie.obj : warning LNK4217: locally defined symbol ??4?$value@H@options@@QEAAAEAV01@$$QEAH@Z (public: class options::value<int> & __cdecl options::value<int>::operator=(int &&)) imported in function "void __cdecl options::tie_setting(class options::value<int> &,class QComboBox *)" (?tie_setting@options@@YAXAEAV?$value@H@1@PEAVQComboBox@@@Z)
1>tie.obj : warning LNK4217: locally defined symbol ??B?$value@H@options@@QEBAHXZ (public: __cdecl options::value<int>::operator int(void)const ) imported in function "void __cdecl options::tie_setting(class options::value<int> &,class QComboBox *)" (?tie_setting@options@@YAXAEAV?$value@H@1@PEAVQComboBox@@@Z)
1>tie.obj : warning LNK4217: locally defined symbol ??B?$value@_N@options@@QEBA_NXZ (public: __cdecl options::value<bool>::operator bool(void)const ) imported in function "void __cdecl options::tie_setting(class options::value<bool> &,class QCheckBox *)" (?tie_setting@options@@YAXAEAV?$value@_N@1@PEAVQCheckBox@@@Z)
1>tie.obj : warning LNK4217: locally defined symbol ??4?$value@VQString@@@options@@QEAAAEAV01@$$QEAVQString@@@Z (public: class options::value<class QString> & __cdecl options::value<class QString>::operator=(class QString &&)) imported in function "void __cdecl options::tie_setting(class options::value<class QString> &,class QComboBox *)" (?tie_setting@options@@YAXAEAV?$value@VQString@@@1@PEAVQComboBox@@@Z)
1>tie.obj : warning LNK4217: locally defined symbol ??B?$value@VQString@@@options@@QEBA?AVQString@@XZ (public: __cdecl options::value<class QString>::operator class QString(void)const ) imported in function "void __cdecl options::tie_setting(class options::value<class QString> &,class QComboBox *)" (?tie_setting@options@@YAXAEAV?$value@VQString@@@1@PEAVQComboBox@@@Z)
1>tie.obj : warning LNK4217: locally defined symbol ??4?$value@Vslider_value@options@@@options@@QEAAAEAV01@$$QEAVslider_value@1@@Z (public: class options::value<class options::slider_value> & __cdecl options::value<class options::slider_value>::operator=(class options::slider_value &&)) imported in function "void __cdecl options::tie_setting(class options::value<class options::slider_value> &,class QSlider *)" (?tie_setting@options@@YAXAEAV?$value@Vslider_value@options@@@1@PEAVQSlider@@@Z)
1>tie.obj : warning LNK4217: locally defined symbol ??C?$value@Vslider_value@options@@@options@@QEBA?AV?$dereference_wrapper@Vslider_value@options@@@detail@1@XZ (public: class options::detail::dereference_wrapper<class options::slider_value> __cdecl options::value<class options::slider_value>::operator->(void)const ) imported in function "public: void __cdecl <lambda_8b45ef41f6e404b8ef95f86c065f9452>::operator()(void)const " (??R<lambda_8b45ef41f6e404b8ef95f86c065f9452>@@QEBAXXZ)
1>tie.obj : warning LNK4217: locally defined symbol ??R?$value@Vslider_value@options@@@options@@QEBA?AVslider_value@1@XZ (public: class options::slider_value __cdecl options::value<class options::slider_value>::operator()(void)const ) imported in function "void __cdecl options::tie_setting(class options::value<class options::slider_value> &,class QSlider *)" (?tie_setting@options@@YAXAEAV?$value@Vslider_value@options@@@1@PEAVQSlider@@@Z)
1>tie.obj : warning LNK4217: locally defined symbol ??4?$value@VQVariant@@@options@@QEAAAEAV01@$$QEAVQVariant@@@Z (public: class options::value<class QVariant> & __cdecl options::value<class QVariant>::operator=(class QVariant &&)) imported in function "void __cdecl options::tie_setting(class options::value<class QVariant> &,class QComboBox *)" (?tie_setting@options@@YAXAEAV?$value@VQVariant@@@1@PEAVQComboBox@@@Z)
1>tie.obj : warning LNK4217: locally defined symbol ??B?$value@VQVariant@@@options@@QEBA?AVQVariant@@XZ (public: __cdecl options::value<class QVariant>::operator class QVariant(void)const ) imported in function "void __cdecl options::tie_setting(class options::value<class QVariant> &,class QComboBox *)" (?tie_setting@options@@YAXAEAV?$value@VQVariant@@@1@PEAVQComboBox@@@Z)
1>opentrack-options.vcxproj -> C:\Dev\GitHub\bin\opentrack-64\options\Debug\opentrack-options.dll
1>Done building project "opentrack-options.vcxproj".
========== Build: 1 succeeded, 0 failed, 2 up-to-date, 0 skipped ==========```

sthalik, 2019-03-03:
Thanks for the report! The debug build now seems to work.

closed 2019-03-03 by sthalik