opentrack/opentrack #908: request: Response Scalling

DukeNewman, 2019-03-06:
Hi sthalik,

been using OpenTrack for a while. Once I started with 3-point-LED Tracking.

For some month I'm using a DIY EDTracker (Arduino Leonardo with MPU9250).

Had some problems with weird movement caused by relative translation (~2.3.10). I think those problems are already tracked and solved. At least no problems since some 2.3.11 beta for me. Great work so far! Beta 3 works fantastic.

My point:
I used the EDTracker GUI (none pro). It's not the smoothest way but it provides one nice feature called "Response Scalling". So one can very easyly change the sensibility of the input data. I used this to get a full 360° (+-180°) turn for the input. So all my OpenTrack-profiles where allways set to 180° input.

Now I swapped to another project for (DIY Head Tracking/RAZOR AHRS) because it allows me to use the OpenTrack Input-plugin for Hatire Arduino with no need of a third programm. Also it's more up to date and offers some other benefits thow (centering). But it lacks some feature like "Response Scalling".

Therefore I'm in the need of changing all the mapping in OpenTrack if I want to change the sensibility!

So my request is, why not implement such a feature in OpenTrack. I'm sure it could be an additional advantage also for other sources of input.

sthalik, 2019-03-07:
What does that feature do?

DukeNewman, 2019-03-07:
As I tried to explain, it ads a scaling factor to the data of the axes.
[Look at this image.](https://www.edtracker.co.uk/images/articles/opentrack/modUICaptures_001.gif) You can see scaling for Yaw and Pitch. Roll is the same as one of the other axis (don't know why -- ask the programmer about this).

If you want more or less sensibility, you easily change the scaling factor without the need of changing anything to your mapping in OpenTrack.

E.g. for my setup, the factor is about 4. So I can get +-180° by turning my head +-45°. For me it's much more easy to setup the mappings for about 180°. By the way: a friend of mine uses the same mappings but with a different scaling.

Hope you get the point.

Slion, 2019-03-09:
Sounds like you could do that with the mapping.

DukeNewman, 2019-03-10:
I don't think so (please correct me If I'm wrong). But maybe I couldn't describe it exactly enough:

I think that you (Silon) assume that I used to use the EdTracker GUI without Opentrack. But instead the EdTracker GUI/driver is preciding Opentrack. So Opentrack eventually get the EdTracker input data (by Joystick plugin) and they already include the scaling factor.

Now without EdTracker preciding, Opentrack gets the input data by Hatire plugin -- always 1:1. And that's the point where I would like to see a new feature in Opentrack to scale the input data before mapping or instead scale the output data just after mapping.

PS: I can very much recommend headtracking by gyro/accelerometer. It only gives you the three rotation axes (you can still use "pseudo-6DOF") but with very high precision for very low cost! It's reliable, light weight and easy to use. I don't realize why this method seems to be so underrated!?

@sthalik:
Maybe you could realize this feature in cooperation with FuraX49 and his Hatire plugin.

sthalik, 2019-03-13:
It's a good feature to have, but I won't implement this immediately. It needs to wait for a larger UI redesign as the change is disruptive. I'm in talks with another party interested in UX improvement.