opentrack/opentrack #910: Add Kinect support to Point Tracker

Slion, 2019-03-09:
Here I'll be tracking progress of Kinect support for Point Tracker.

sthalik, 2019-03-14:
Try implementing something resembling `tracker-pt/module/camera`. Then it's a matter of replacing usages of that struct with a pure virtual class. That I can do on my own, including registering camera providers from separate modules.

Slion, 2019-03-17:
I got a first working draft with minimal changes to Point Tracker.
* If Kinect hardware is detected it adds a 'fake' camera option.
* I implemented a `pt_camera` object for Kinect IR Sensor.
* I've delayed the creation of the traits object which notably defines the implementation of the camera until the settings can be checked.
* If the Kinect IR Sensor camera is selected a traits object using the Kinect camera is used.

Unfortunately the point extractor seems to be expecting an RGB-888 buffer so the Kinect camera object does convert its 16-bits single channel buffer to an 8-bit three channels. It's not such a big deal though it could be nice to optimise that at some point.

Now I have to configure Point Tracker which is not exactly straight forward. I never used Point Tracker with any other hardware. Right now I'm still getting strange results. The three points on my cap are detected nicely but their interpretation is somewhat odd. After adjust the diagonal field of view as specified by the Kinect API I'm getting even stranger results.

Slion, 2019-03-17:
It looks like Point Tracker is working fine at first but then if I get too close to the sensor it detects a lot of noise and even after I went back into a proper position it keeps providing broken yaw and roll around 180 degree. Pitch appears to be broken too. The only way to get it working again is to restart the tracker.

Slion, 2019-03-17:
> The only way to get it working again is to restart the tracker.
Actually doing a recenter is enough. Still that's rather annoying. However it's out of scope of this issue.

Slion, 2019-03-17:
After configuration and testing I'm happy with this implementation. @sthalik I'll be  looking forward to your feedback and code review hopping we can contribute it to master.

One thing on my task list though is to workout how to provide image dimension and fov from the code without requiring the user to fill that in the settings.

sthalik, 2019-03-18:
> if I get too close to the sensor it detects a lot of noise and even after I went back into a proper position it keeps providing broken yaw and roll around 180 degree

Any persistent breakage I can only account for if you select the "dynamic pose resolution" checkbox for cap tracking. If you're tracking a cap, centering will reset the internal state either way. If there's breakage outside this scenario, I'd like to know how to reproduce it, and in particular,

- whether you use automatic thresholding
- what slider position do you use

sthalik, 2019-03-18:
The changes you made are ok generally but done at the wrong level of abstraction, architecturally speaking. The idea behind a modular design like opentrack's is to avoid mixing dependencies.

I'm going to make some ground work so that you can plug in your camera from an arbitrary module.

Slion, 2019-03-18:
> The idea behind a modular design like opentrack's is to avoid mixing dependencies.

I can see why you would not want to add even an optional dependency against Kinect to Point Tracker. Though to me the easiest path forward would be to merge to master a version of what I have and improve the architecture from there. I also don't mind waiting for your architecture changes and re-implement Kinect Point Tracker on top of it. It should be straight forward anyway since the core of the work is just that `CameraKinectIr` object.

Slion, 2019-03-18:
> I'd like to know how to reproduce it

My issue is really easy to reproduce with Kinect. Just create some noise/saturate the IR buffer by moving close enough to the sensor for instance and then get back into position. More often than not both yaw and roll are stuck around +/-170 until you reset/center, that's even though all three point appear to be found properly. I don't have other compatible IR camera I could test that with so it could be that this is very specific to Kinect. We don't happen to support the TrackIR camera do we?

I'm using the cap model with a TrackIR construct. I found that without "dynamic pose resolution" the tracker breaks as soon as the top point is not in between the left and right ones anymore i.e. whenever you turn your head too far left or right.

Slion, 2019-03-18:
What kind of CPU usage do you usually get with Point Tracker. Mine was around 15% on an Intel Core i7-7700K 4,2 GHz. Then again I'm doing like two buffer copies in the `CameraKinectIr` implementation alone so we have clearly some room for optimisation there.

Next I'll probably be looking at implementing support for 16-bits single channel buffer in `PointExtractor`.

Slion, 2019-03-18:
Actually I have access to a PS3-Eye camera so I'll try to reproduce that issue with it. Moreover if the issue is caused purely because of the saturation I could try to reduce the range when converting from 16-bits to 8-bits. See that call to `Mat::convertTo` and use arbitrary `min` and `max` value when computing the scale factor.

Slion, 2019-03-18:
I could workaround that issue by adding some auto reset feature when the angles do not make sense.

sthalik, 2019-03-18:
> Next I'll probably be looking at implementing support for 16-bits single channel buffer in PointExtractor to tackle that problem.

It should be easier with the new camera API. Add a "depth" element, signifying byte count per pixel. There's already an element called "channels" but it's not fully implemented (e.g. the preview) and can crash in opencv's innards.

Keep in mind that you can add a camera from *any* module, including another tracker. Use `OTR_REGISTER_CAMERA`.

There are some issues though:

- there are many assumptions about 8-bit color depth
- operations on `uint16_t` are slow
- it uses half the SIMD operands
- the Aruco tracker needs adjusting too. you needn't touch the Wii tracker though, it doesn't use the camera API.

> I could workaround that issue by adding some auto reset feature when the angles do not make sense.

I have my reservations here. There are 6 possible solutions of which only 1 is correct.  They have varying degrees of making sense. If you go for it, try very high yaw, pitch, and roll values as a heuristic.

> Just create some noise/saturate the IR buffer by moving close enough to the sensor for instance and then get back into position. 

Is this a typical use-case?

> What kind of CPU usage do you usually get with Point Tracker.

Be sure to set a constant CPU clock rate for reliable testing. I get 10% total system load with the main window minimized, but some 80% is spent in the driver, including the kernel. Use VTune to measure it, there's a community license.

sthalik, 2019-03-18:
Regarding the Aruco tracker, there can be some enum signifying whether custom depth is allowed. If the tracker is ignorant, punt and provide 3-channel 8-bit buffers. You can work with the camera API, specifically the `make_camera` member function and the `video::make_camera` function.

sthalik, 2019-03-18:
`can_show_dialog` isn't used yet, but please implement it as well. It probably returns `false` in all cases in your impl. I'm going to hook it up to the dialog later.

Slion, 2019-03-18:
> I have my reservations here. 

So do I. It's just a workaround really and ideally we should identify the problem in the core algorithm. However until someone invest the time to look into it should make things work. More testing needed.

> Is this a typical use-case?

Unfortunately yes. You may be stretching or scratching your forehead. Even walking away and coming back to your desk can trigger the issue. Moreover I was just giving a sure example of how to reproduce, it can happen in many other ways. Having to manually recenter every 2 minutes is just a pain.

> I get 10% total system

Alright so I guess the extra 5 I was seeing are probably accounted for by the extra buffer copy and possibly Kinect APIs themselves.

Slion, 2019-03-18:
> Regarding the Aruco tracker

I'm not even building that guy. Should I?

Slion, 2019-03-18:
I implemented the new framework but I've got some issues with it.
As soon as getFrame returns false the framework stops and does not try requesting new frames.
Also the fact that it was stopped is not reflected in the UI.
If getFrame returns true with a null frame data it crashes.
IMHO the framework should not stop requesting frame as soon as one fails.
It should just skip it until it succeeds.

Slion, 2019-03-18:
https://github.com/opentrack/opentrack/pull/new/kinect-point-tracker-v2

Slion, 2019-03-19:
It would be great if the camera implementation could specify its specs so that they are populated in the UI rather than needing the user to do that. (fov, width, height, fps? ...)
The `frame` object should have a `channelSize` property so as to open the path the supporting 16-bits.

Slion, 2019-03-19:
I tried setting up a PS3 Eye hardware with Point Tracker but I have not idea how people are getting this to work. I'm thinking you need IR LED rather than reflectors like I have. I'll just give up on that for now.

sthalik, 2019-03-19:
> So do I. It's just a workaround really and ideally we should identify the problem in the core algorithm. However until someone invest the time to look into it should make things work. More testing needed.

I'm not confident that the algorithms have a bug per se. Rather, "garbage in, garbage out". Anyway use `PointTracker::reset_state()` in your reset heuristic. 

> I implemented the new framework but I've got some issues with it.
> As soon as getFrame returns false the framework stops and does not try requesting new frames.
> Also the fact that it was stopped is [...]

As you can see in `Tracker_PT::run` and `Camera::get_frame_`, failing to get a frame doesn't result in a failure. It's only gonna busy-loop if there's no frame for any subsequent calls.

The code assumes the `struct frame` data to be initialized if the second element of the tuple is `true`.

> The frame object should have a channelSize property so as to open the path the supporting 16-bits.

Fixing all places where `unsigned char` is assumed will be painful. You're even going to have an overflow in meanshift in the point extractor given limited range of `float` and taking a square for each pixel.

Given the amount of changes needed for PT to support alternate color depths I'm on the fence whether changing this assumption would be too invasive for the PT codebase. I doubt changing the depth would increase accuracy by a visible margin. It's like the mouse DPI gimmick a bit.

> It would be great if the camera implementation could specify its specs so that they are populated in the UI rather than needing the user to do that. (fov, width, height, fps? ...)

Ignore the parameters you don't use. Later we can think of something.

> I tried setting up a PS3 Eye hardware with Point Tracker but I have not idea how people are getting this to work. I'm thinking you need IR LED rather than reflectors like I have. I'll just give up on that for now.

Do you have the IR filter removed?

sthalik, 2019-03-19:
> As soon as getFrame returns false the framework stops and does not try requesting new frames.

Can you track this down please? `tracker/pt` seems to ignore missing frames.

sthalik, 2019-03-19:
To elaborate on color depth.

See that BT. 709 has truncation in [luma coefficients](https://en.wikipedia.org/wiki/Rec._709#Luma_coefficients). Naturally, one could resample the 8-bit frame to 16-bit or even a floating-point format in order to avoid the truncation. But this is never done given the effect is so insignificant.

The image frame, whether PS3 Eye or a regular consumer webcam, has lots of noise even with a sane gain value. The noise is in the bounds of some 4/256 if not more. You should probably be able to see the noise from Kinect's sensor when stretching the video feed to take your whole desktop.

Slion, 2019-03-19:
>  failing to get a frame doesn't result in a failure

 I think that's only during start-up which is fair enough I guess.

Slion, 2019-03-19:
> Do you have the IR filter removed?

No I did not. I borrowed it so I figure I should try not to break it :)

Slion, 2019-03-19:
To me the main point of 16-bits support is to improve performance by avoiding frame buffer manipulation and copies.
Actually first I think we should implement support for 8-bits single channel, that should be fairly easy.
You see the last thing the Kinect camera does is convert from grayscale to RGB and the first thing Point Tracker does is convert it back from RGB to grayscale. Maybe it's working already I have not tried it within the new framework.
Then we could look into 16-bits support. However all of that is outside the scope of this issue.
I will focus on contributing this implementation with the existing framework back to master and then we could think of improving it.

Slion, 2019-03-19:
https://github.com/opentrack/opentrack/commit/bfb86f510e30fe69355dc5ca084459e96867c34f#r32817554

Slion, 2019-03-19:
One more thing about PS3 Eye. It does work with 2.3.11 BETA3 but it's not listed when I use my own 64-bits debug build.

sthalik, 2019-03-19:
The closed driver's filter only has a 32-bit version.

sthalik, 2019-03-19:
> You see, the last thing the Kinect camera does is convert from grayscale to RGB, and the first thing Point Tracker does is convert it back from RGB to grayscale.

Converting from RGB to grayscale isn't necessary if PT is made to support it. Then Kinect has the same overhead as a regular webcam.

You'd need to modify `PointExtractor::color_to_grayscale` to do a fast exit if the frame is single-channel. Then go for `Preview& Preview::operator=(const pt_frame& frame_)` and convert to RGB there.

Slion, 2019-03-19:
I gave this a try with the auto reset feature in ED and it's very usable. Though it could take a while to fine tune the mapping and filtering. It's very close from the TrackIR experience.

closed 2019-03-19 by Slion