opentrack/opentrack #911: Slow to start / crash

neyz, 2019-03-10:
I saw a similar issue #782 but not sure it's related since my dll list looks clean.

Program takes more than 1 minute to start and then when i press start it will just crash "not responding"

Using version 2.3.11 beta3
Windows 10 1809 French
Oculus rift

I use this computer as a gamestream server but i disabled the option in the nvidia panel.

List of dll sorted by path in attached file.

Tell me if i can do anything else !

Thanks
[opentrack.exe.txt](https://github.com/opentrack/opentrack/files/2949460/opentrack.exe.txt)

sthalik, 2019-03-14:
Please use Task Manager to create a minidump and send it here as an archive (copy-paste the file in the text box). Here's the whole procedure: <<https://support.microsoft.com/en-us/help/931673/how-to-create-a-user-mode-process-dump-file-in-windows>>. You specifically want to use "method 1".

neyz, 2019-03-21:
Sorry didn't see your reply there. I just created a dump but it's 70MB and github won't have it. Here is a shared link with the zipfile containing the dump : https://drive.google.com/file/d/1yVEaodw3xERy6x0DCY5A7oS0SlhcC6Ya/view?usp=sharing

sthalik, 2019-03-21:
It's more likely that I'd be able to help you with the Oculus Rift crash. Please use today's (20180321) build from <<http://opentrack.misaki.pl>>.

I can see from the dump that the program has basically started and is processing input events. I can only assume that the error lies deeper within the Qt library. Are you using multiple displays? There used to be a problem like that more than a year ago but it's believed fixed. I should be able to offer a workaround if that's the case.

Also, see that opentrack listens for keyboard and joystick input by necessity and that can trip up antivirus programs. That behavior is essentially identical to a keylogger but there's no separate mechanism that opentrack could use. It's fine with Windows Defender but some antivirus programs on the market can cause problems, and caused them in the past. See if turning off realtime protection during program startup helps any.

neyz, 2019-03-21:
Thank you for the quick answer, I just tried the new build and everything works fine with that one ! 

- I don't have any antivirus except windows defender.
- I do have a second detected display in windows but i think it's the rift being detected (listed as VR desktop)

Let me know if you need me to do any post actions.

thanks !

sthalik, 2019-03-22:
That's great to know! I'm inclined to believe that the issue was caused either by the build system, or something specific to your environment. There's nothing in the code per se that needs changing.

closed 2019-03-22 by sthalik

neyz, 2019-03-22:
Well it seems i spoke too soon as i have the exact same problem again.

I shutdown all applications / services and still have the issue. It worked fine the other day so you must be right that it comes from my environment but this is a clean install, i'm at a lost.