opentrack/opentrack #912: Issue when going from positive to negative 180 yaw

UnknownEvil, 2019-03-17:
Instead taking the quickest path from 180 to -180 (which is not moving) it makes a full rotation. Really unpleasant

sthalik, 2019-03-18:
What output is it?

sthalik, 2019-03-28:
I'm not sure if the problem is located in the filter. Can you please see if disabling the filter fixes the issue? I'd know which component has the issue at least.

UnknownEvil, 2019-03-28:
Accela causes the full rotation. No filter causes the yaw to lock to +/-180 once I pass +/-180, respectively. Kalman seems to almost work but it appears to face the opposite direction for a few frames before returning to the correct direction. EWMA does something like the no filter and the Kalman combined.

sthalik, 2019-03-28:
Let me get this clear, does disabling filtering result in proper behavior?

UnknownEvil, 2019-03-28:
No. 

> No filter causes the yaw to lock to +/-180 once I pass +/-180, respectively. 

With the filter disabled if I rotate my phone to -180 yaw (to the left), the output yaw will lock to -180. If I rotate it to 180 yaw (the right) the output will lock to 180.

If I look left and it locks to -180, I can look right and it will still work until locking to 180. |
I don't know if it's intended behavior for yaw to max out at 180 and -180?

sthalik, 2019-03-28:
@UnknownEvil please see if it works at least with no filter in the current build. Once it works with no filter I can fix Accela.

Builds as usual at <<http://opentrack.misaki.pl>>.

UnknownEvil, 2019-03-31:
Same thing as the thing I said. No filter causes the yaw to lock to +/-180 once I pass +/-180, respectively.

sthalik, 2019-04-01:
I see the issue's caused by centering. Please see today's build.

closed 2020-01-27 by sthalik