opentrack/opentrack #913: Freepie IMU stops sending tracking data after running in the background after a while

simbake, 2019-03-18:
Hi,
Opentrack tracking pauses after running freepie imu in the background. The only way to resume tracking is to open freepie imu. This happens on Samsung s10e android 9.
Please help.

sthalik, 2019-03-18:
Disable battery saving mode for this application.

simbake, 2019-03-19:
Is it under optimize battery usage? Please guide me on how to disable. Thanks

simbake, 2019-03-19:
I turned off battery optimization for the app but the issue still occurs.

sthalik, 2019-03-19:
I haven't had any Samsung phones in a while and can't help, sorry. Maybe somebody else knows.

quietbug, 2019-06-26:
I got exactly the same issue after flashing Android 9. `dumpsys deviceidle disable` also not resolve it. Redmi 5+.

sthalik, 2019-06-27:
I think this is related to general optimizations in newer Android versions.

simbake, 2019-06-27:
It would be nice to see and update of the IMU.

diego123456789, 2019-06-29:
same problem here with Motorola -> Android 9

Ardemon96, 2019-08-30:
Me too - redmi note 7, android 9

sthalik, 2019-09-22:
Unless there's a way to disable all battery life optimizations the freepie tracker has to be discontinued unfortunately. :(

siriux, 2019-09-24:
I ported the freepie imu to newer android versions for an app I'm playing with and it works without pauses even on the background and I think I remember even with the screen locked.

It has a different name (WishIMU) and it's on a different package because I've use it as a base for something different, but I still keep the first copy of the code that was a straight port and compatible with freepie imu.

Keep in mind this is old code, not very well tested, and it might contain plenty of bugs, but it seemed to work.

I attach the code and a debug apk I have. You can use WishIMU and the code under the original MIT license (I added my name, it would be nice if you keep it) as a replacement for freepie imu, or move it to the old package, or just backport the changes (not straight forward, everything is really old, that's why I created a new package in the first place).

[WishIMU Compatible With FreePIE.zip](https://github.com/opentrack/opentrack/files/3645831/WishIMU.Compatible.With.FreePIE.zip)

[com.wishsalad.wishimu-debug.apk.zip](https://github.com/opentrack/opentrack/files/3645835/com.wishsalad.wishimu-debug.apk.zip)

Let me know if it works for you.

simbake, 2019-09-24:
Hi Siriux, thanks for the update. Unfortunately it doesn't work for me, it seems to not be able to detect my sensors as when I click on 'Debug' there are no values and Opentrack doesn't receive any values. I have tested with the old version also and this sends values and detects the sensors.

Morblax, 2019-09-24:
Thank you Siriux, I merged your sources with original ones and now my Galaxy Note 8 send data to FreePIE ! I don't test it on Opentrack, it's dirty, but it works.

The builded APK :
[FreePIE_IMU_Android9.zip](https://github.com/opentrack/opentrack/files/3649638/FreePIE_IMU_Android9.zip)

siriux, 2019-09-25:
@simbake The debug doesn't work in my code (I didn't need it, so I didn't port the code) I forgot to mention it. You can try with FreePIE, to see if it works.

@Morblax Good to know that it works for you. I tried with opentrack some time ago and it worked for me, but as I said, this code is something old I didn't test too much.

simbake, 2019-09-25:
@Morblax Thanks for sharing. Your version works for me.
@siriux Thanks for sharing and updating the tool, Morblax version works on Opentrack haven't tried FREEPIE as I dont use it.

sthalik, 2019-10-12:
@siriux can I bundle it with opentrack?

siriux, 2019-10-14:
@sthalik Sure, you can bundle it, port it, or do whatever you want with it. 

It would be nice if you keep the MIT license file with my name (along the original authors).

sthalik, 2020-01-27:
Added to the stable version. Thanks!

closed 2020-01-27 by sthalik

MABlosfeld, 2021-11-13:
I'm using RESHADE/MOONLIGHT/FREEPIE IMU/OPENTRACK apps to play with VR HEADSET on SMARTPHONE. I did several tests and found that every time I turn on moonlight, it turns freepie off. Is there any chance that moonlight will allow other apps to run in the background?

r57zone, 2021-11-14:
MABlosfeld, have you tried the new FreePie IMU version?

MABlosfeld, 2021-11-14:
I looked over the version installed on my phone and it says VERSION 1.0 and the file name description is FREEPIE_IMU_ANDROID9.APK
I did other tests and every time I turn on an app inside GAME LAUNCHER SAMSUNG, the freepie turns off by itself.

MABlosfeld, 2021-11-14:
Perhaps the solution is to include all the FREEPIE source code within the MOONLIGHT source code, this way, when turning on MOONLIGHT it will automatically transmit data from the motion sensors to OPENTRACK or any other application.

MABlosfeld, 2021-11-14:
Perhaps a small financial contribution is needed for the development of updates and new projects. OPENTRACK/FREEPIE/MOONLIGHT/RESHADE are 4 excellent projects and each one has its due value. I support the idea of ​​creating a way to contribute.