opentrack/opentrack #916: Point Tracker to support single channel frame buffer

Slion, 2019-03-20:
It seems Point Tracker does not support single channel frame buffer which is in fact the most common format for IR cameras. Kinect IR camera implementation notably needs to convert from grayscale to RGB to make Point Tracker happy, thus degrading performance.

Implementation should be fairly straight forward. We will need to consider how that affects our "Color channel used" parameter. Should we just ignore it when single channel buffer is provided? I guess so.

[See discussion there](https://github.com/opentrack/opentrack/issues/910#issuecomment-474471388).

sthalik, 2019-03-20:
Yes, ignore color channel settings it in the point extractor. Convert to RGB only in the preview frame, use as-is in the extractor.

I made some comments for #914. Please address these as well.

Slion, 2019-04-24:
This won't be needed for now, since we determined Kinect IR camera would not work properly together with Point Tracker because its implementation is making a bunch of assumption about your camera specs being somewhat similar to PS3 Eye.

closed 2019-04-24 by Slion