opentrack/opentrack #917: ITracker::center not called when starting a tracker

Slion, 2019-03-20:
ITracker::center should be called by the pipeline when starting a tracker cause we effectively center.

sthalik, 2019-03-21:
The intention is to notify the tracker that user-initiated center was performed. It's not meant to be called merely because the pipeline defaults to an identity pose.

Slion, 2019-03-22:
Though that would be useful and more consistent. In fact I was hoping to implement solution where when no valid position is found it defaults to the center.

sthalik, 2019-04-07:
> user may want to take his hat off for a while and still play the game.

Unfortunately that's not very feasible. The freetrack protocol library doesn't disable tracking on its own despite protocol support existing. The dll has been whitelisted by anti-cheat engines (some unknown to me). The game will perform as if it's always perfectly centered with no way to perform mouse panning.

>  In fact I was hoping to implement a solution where when no valid position is found it defaults to the center.

It doesn't work in practice. I broke it that way by accident and it was unusable.

Slion, 2019-04-13:
>  The game will perform as if it's always perfectly centered with no way to perform mouse panning.

I did not think of that. Though in my case that should not be much of a problem.

sthalik, 2019-04-14:
Try adding an `else` branch to `if (success)` in PT and see how bad it performs.

sthalik, 2019-04-14:
Basically you're gonna get intermittent failures.

Slion, 2019-04-27:
I'm implementing auto center in Easy Tracker and that works just great, much like TrackIR in fact. One second after taking off your hat it centers. I've not tested trying to move your head with other controls but I don't see why it would behave any different than it does when you are wearing your hat anyway.

sthalik, 2019-04-28:
> It would be nice to have this bug fixed to have all case covered but in real life it's just never a problem since user will pretty much always explicitly request center operation anyway.

It's still a pain for cap users since intermittent failures make it reset on its own. The failures are more common on rotated frames where a single LED won't show fully.

Also between putting on the cap and centering, the ingame camera stays in a random badly-rotated pose.

Slion, 2019-04-28:
It needs more testing but, so far, the Easy Tracker auto centering implementation behaves just like TrackIR with no downside I can think of.

closed 2020-01-27 by sthalik