opentrack/opentrack #920: How to use OpenCV in Debug on Windows?

Slion, 2019-03-27:
When using OpenCV DLLs on Windows I get a crash in Point Tracker cv::putText.
That's true for the official OpenCV distribution and when compiling it myself.
It seems the std::string parameter is not passed properly to the DLL, different calling convention?

Therefore I built OpenCV as a static library to use it thus in OpenTrack. Unfortunately I then hit a linker problem in OpenTrack which looks like this:
`Error	LNK2038	mismatch detected for '_ITERATOR_DEBUG_LEVEL': value '2' doesn't match value '0' in camera.obj	opentrack-tracker-pt	C:\Dev\GitHub\bin\opentrack-64\tracker-pt\module\opencv_imgproc410d.lib(resize.obj)`

Both OpenCV and OpenTrack used CMake generator `Visual Studio 15 2017 Win64`.
Any clue what could be going on?
Any chance getting instructions on how to build OpenCV for debug?

sthalik, 2019-03-27:
Remove that line: https://github.com/opentrack/opentrack/blob/c90f8fbd1a83ffeed6b0bb9d55e91f4a4a9b8641/cmake/opentrack-platform.cmake#L127

sthalik, 2019-03-27:
But it's a bad idea, the iteractor CRT feature is a CPU hog. Better add it to the opencv build system.

sthalik, 2019-03-27:
You shouldn't have any issues with using `-Zi`/`-DEBUG` for release builds. See `cmake/msvc.cmake` for my toolchain with tons of optimizations. Some variables get clobbered but the debugging is very productive.

Slion, 2019-03-27:
Thanks, that fixed my build. Also `cv::putText` is not an issue anymore when using OpenCV static libraries.

What about moving it to your optimised toolchain so that our default build works with the default OpenCV build?

closed 2019-03-28 by sthalik

sthalik, 2019-03-28:
> Thanks, that fixed my build. Also cv::putText is not an issue anymore when using OpenCV static libraries.

> What about moving it to your optimised toolchain so that our default build works with the default OpenCV build?

If you use any static dependencies, please use my static toolchain file. You'd see the difference in the size for the "modules" subdirectory, not counting the Kinect files of course.

PR #922 (2019-03-28): Using standard _ITERATOR_DEBUG_LEVEL by default
Thus making default MSVC build working with default OpenCV static library build.
Closes #920.
commit 905bd334 (2019-03-28): Not changing _ITERATOR_DEBUG_LEVEL default value any more when using default toolchain.
Thus making default MSVC build working with default OpenCV static library build.
Closes #920.