opentrack/opentrack #926: PointTracker on Linux?

waterlubber, 2019-04-03:
I've installed Opentrack on Arch Linux through the AUR package: https://aur.archlinux.org/packages/opentrack/

I have it working and it seems to be working fine. However, PointTracker does not appear in the list of input trackers. I have two working webcams on the system (/dev/video0 and /dev/video2) There is no notable output when running it in console. 

How to I add the Pointtracker plugin? Should I just run this inside of WINE for use with Elite Dangerous?

sthalik, 2019-04-04:
> Edit: It appears that the tracker-pt file is missing from the /usr/libexec/opentrack/ directory. How do I compile or get this? I can't get Opentrack working in WINE. (the webcam doesn't work)

Make sure to set `OpenCV_DIR` in cmake cache.

waterlubber, 2019-04-04:
It appears to be set correctly (in the CMakeCache.txt file in the cv/ directory), there is the following line:
`OpenCV_DIR:PATH=/usr/lib/cmake/opencv4`
and that directory has the files in it:
```
/usr/lib/cmake/opencv4/OpenCVConfig-version.cmake
/usr/lib/cmake/opencv4/OpenCVConfig.cmake
/usr/lib/cmake/opencv4/OpenCVModules-release.cmake
/usr/lib/cmake/opencv4/OpenCVModules.cmake
```
I then ran cmake with this command (which is almost certainly wrong):
`OpenCV_DIR=/usr/lib/cmake/opencv4 cmake .. -GNinja -DCMAKE_BUILD_TYPE=Release -DSDK_WINE_PREFIX=/home/waterlubber/.wine -DCMAKE_INSTALL_PREFIX=~/.local/share -OpenCV_DIR=/usr/lib/cmake/opencv4`

Cmake gave the following output: https://pastebin.com/xXwuBW5S
and then ninja. Ninja failed, with the following output: https://pastebin.com/EBNUCkV6

Are there any building instructions for this? I'm not very familiar with cmake at all and am generally just running random console commands at this point.

waterlubber, 2019-04-05:
So I've tried a few other things (compiling in different ways with and without OpenCV) but I still can't get it to sucessfully compile with OpenCV. If it makes a difference, I'm using OpenCV 4.0.1-6, should I use an older version like 4.0.0, 3.4.5, or 2.4.13?

sthalik, 2019-04-05:
That error is unrelated to opencv. Pull from unstable and try again.

That pkgbuild should bump `_pkgver`.

sthalik, 2019-04-05:
`-fpermissive` shouldn't be required under any circumstances. It should be fixed now.

If you're having problems with loading the app after that, I have some instructions.

waterlubber, 2019-04-05:
Alright, I just compiled after cloning with:
`mkdir build`, `cd build`, `cmake ..`, `make`, and `make install`. I set LD_LIBRARY PATH to `build/install/libexec/opentrack` (where all the libraries were) and ran it. I still did not see the video options:
[image]

Terminal output was as follows:
``` waterlubber@gamingPC  ~/.local/share/src/opentrack/build/install/bin   unstable  ./opentrack 
[/home/waterlubber/.local/share/src/opentrack/qxt-mini/qxtglobalshortcut.cpp:64]: qxt-mini: adding event filter
[/home/waterlubber/.local/share/src/opentrack/qxt-mini/qxtglobalshortcut_x11.cpp:349]: qxt-mini: fixed x11 autorepeat```

I also tried the opentrack-git AUR package, which downloads from the repo, and it did not work either.

sthalik, 2019-04-06:
@waterlubber please show the output of

```
build/ $ grep OpenCV_DIR CMakeCache.txt
```

sthalik, 2019-04-06:
Thanks for noticing this, especially so soon after the mistake was made :+1:. I was able to reproduce this bug and fix it.

closed 2019-04-06 by sthalik

sthalik, 2019-04-06:
@waterlubber,
>  should I use an older version like 4.0.0, 3.4.5, or 2.4.13?

Any version above 4.0 or some newer 3.x should work. I can't guarantee 2.x will work for all modules.

waterlubber, 2019-04-06:
Thank you for all the help!
I can see and configure the PointTracker option, but now I'm experiencing a different issue (#930).