opentrack/opentrack #928: 2.3.11 - beta overwrote my other instance of OT.

Iknowlongnamerightwhatwasithinking, 2019-04-04:
So I had two  instances of OT installed in separate folders.
2.3.9 and 2.3.10.
2.3.9 was to track a second camera. 

Installed 2.3.11-beta to see if it would help with a tracking issue I seem to
have with 2.3.10 and it overwrote 2.3.9 ... 

So um, yeah, all the work I did to get two instances of OT working on my pc went 
up in smoke darn it. 

It's time like this I really wish I had listened to what my mother had said to me. -Arthur. 
Oh yeah,  what was that? -Ford. 
I don't know I never listened to her! -Arthur. 

A.F.

Len62, 2019-04-04:
You installed from the setup.exe ?

Iknowlongnamerightwhatwasithinking, 2019-04-05:
Yes. 
and 2.3.11-beta stopped working. Had to uninstall and go back to 2.3.10. 

Also I wish that programs did not use my documents to use as storage for files. 
Can't they just use the programs own folder. It's such a pain in the neck when 
programs are uninstalled and yet leave folders dotted around the place. 
and why windows let programs install stuff into the windows  folder is beyond me. 

Slight rant over. 

2.3.10 is still working fine with point tracker and freetrack output. 

I still have no idea how to set the model up for my custom head LED setup. 
Have tried everything but it's always just crap. So I left the cross at the top LED 
on the 3LED hat IR setup. and moved on. 
A really in depth explanation with diagrams would be very very useful. I just don't understand how the thing works to do this. : (

A really good explanation of the whole thing would be great. 
The point tracker input and the freetrack 2.0 enhanced output. 
( As this is generally where peeps with IR LEDS etc get stuck...) 

S! 

R.A

Len62, 2019-04-05:
The safest way to have multiple installs, is to download the zip file, and unzip to a folder with the same name as the version. This doesn't fix the issue of the ini's being in the docs folder, but you can have as many installs as you want. Of course you will have to create your own program shortcuts, I give the shortcuts the same name as the version. You do have to be careful with loading the ini files as older ones sometimes do not work with newer versions of OT. This is how my folder structure looks:

C:\Users\lfend\Downloads\opentrack-2.3.11-beta3
C:\Users\lfend\Downloads\opentrack-2.3.10
and so on.  At times I have had as many as five versions installed this way. 

As far as the setup goes, have you entered the dimensions of your custom cap?

sthalik, 2019-04-06:
> The safest way to have multiple installs, is to download the zip file

Word. The installer does nothing except extract the zip file and register an uninstaller and a start menu entry. Nothing at all.

Iknowlongnamerightwhatwasithinking, 2019-04-06:
I am a bit confused here.

C:\opentrack 2.3.9\ was my 2.3.9 opentrack folder. As you might expect. 

When I installed 2.3.11 exe it overwrote the above folder.  
As it was installing it told me the above was the target dir. 
It had already started so I let it finish and yes 2.3.9 was now gone burger. 
It did not ask where I wanted 2.3.11 installed.

Question: 

What is the actual difference of  the  xyz.exe vs xyz.zip ??? 
Is it merely the choice of expander? or is there some other difference???

S! 

R.A.

Iknowlongnamerightwhatwasithinking, 2019-04-06:
ok. Just experimenting. 

If I run opentrack-2.3.10-win32-setup.exe onto a clean machine,
the installer will ask me where do I want it to live. : -)

so I say here: 
c:\opentrack 2.3.20

If I rename that folder to this:
c:\opentrack 2.3.10 folder 1       
and then run opentrack-2.3.10-win32-setup.exe again, it will make a folder:
c:\opentrack 2.3.10 
and install into that. 
Without the - where do you want to plant your OT folder - prompting. 

so I end up with:

 c:\opentrack 2.3.10 folder 1 
 c:\opentrack 2.3.10 

( both of these will load the same .ini file in the user/document/opentrack folder. ) 

If I do not rename the folder and/or leave it in it's original location then  opentrack-2.3.10-win32-setup.exe will over write. 

So 2.3.11 installed over the top of my original 2.3.9 . 

Maybe a way to make sure  you really want to over write your existing OT folder...

I do like the portable OT folder idea with the portable.txt trick to create the ini folder. 
As I recall I had to use 2.3.9 because it was the only version that would work with the single 
point input plugin. and it had to use the user\opentrack\default.ini file.... 

well I think I am done. Time to get out of bed and face the world. Brrrr it's cold! 

S! 

R.A.

sthalik, 2019-04-25:
> Maybe a way to make sure you really want to over write your existing OT folder...

Use the portable version.

Iknowlongnamerightwhatwasithinking, 2019-05-11:
I gave up on that project so it's not an issue.

closed 2019-05-11 by Iknowlongnamerightwhatwasithinking