opentrack/opentrack #930: "protocol dylib load failure" on Start

waterlubber, 2019-04-06:
You may remember me from such issues as #926. I'm on Arch Linux and am experiencing the following error whenever I try to click the button to "Start" tracking:
[image]
```
 waterlubber@gamingPC  /usr/local/bin  ./opentrack 
[/home/waterlubber/.local/share/src/opentrack/qxt-mini/qxtglobalshortcut.cpp:64]: qxt-mini: adding event filter
[/home/waterlubber/.local/share/src/opentrack/qxt-mini/qxtglobalshortcut_x11.cpp:349]: qxt-mini: fixed x11 autorepeat
[/home/waterlubber/.local/share/src/opentrack/logic/runtime-libraries.cpp:25]: protocol dylib load failure
```

This error occurred in three different installations:
- Arch Linux [opentrack-git](https://aur.archlinux.org/packages/opentrack-git/) package
- Compiling with `cmake ..`, `make -j12`, and `make install` (and then running it from the install directory with LD_LIBRARY_PATH set)
- Compiling with `cmake -DCMAKE_INSTALL_PREFIX=/usr/local ..`, `make -j12`, and `sudo make install`, then running from /usr/local/opentrack/bin`. 

For whatever reason, it appears to be looking for a library in the source directories (?) and all I could find about "dylib" is that it's a Mac OS thing. 

[Output of cmake](https://pastebin.com/96B3xm4d)
[Output of make -j12](https://pastebin.com/2R3abtPK)
[Output of make install](https://pastebin.com/LHHZGT1V)

The program doesn't crash, but it doesn't work either. The issue occurs no matter what tracker I have set (Pointtracker, test, UDP input, etc.)

sthalik, 2019-04-06:
You don't have any output mode selected on the bottom left. Is the list empty?

waterlubber, 2019-04-06:
Oh, that was the problem, I forgot to select an input. My bad.

It *is* missing the WINE support that I noticed before, but now it's working fully. 
Thanks for all the help.

sthalik, 2019-04-06:
Wine works if you enable it in the cmake cache.

waterlubber, 2019-04-06:
Hmm, I'm trying to compile with Wine support (I tried `-DSDK_WINE_PREFIX=/usr` and -DSDK_WINE_PREFIX=/`, as well as the set() command in CMakeLists) but I get the following error from CMake:
```
CMake Warning:
  Manually-specified variables were not used by the project:

    SDK_WINE_PREFIX
```
and no WINE support when I compile.

sthalik, 2019-04-07:
Use `ccmake` or `cmake-gui` for this. The variable `SDK_WINE_PREFIX` you can delete, now there's a boolean in its place.

waterlubber, 2019-04-08:
I got it compiled and working fine now, thank you.
Do you know if I need to specify the wineprefix for the game it interfaces with? I'd like to get it working with Elite Dangerous. It does work with joystick emulation, but the Freetrack protocol is a little more responsive and supports roll.

sthalik, 2019-04-08:
> Do you know if I need to specify the wineprefix for the game it interfaces with?

Set it as the environment variable, as usual.

waterlubber, 2019-04-08:
Alright, I'll test that when I can. As of now, I can only get the game to run in Proton, which doesn't play well when sharing a prefix with WINE. Once I can get the game running in WINE, I'll let you know, but until then joystick emulation will suffice. Thanks for all the help, you've been very patient.

closed 2019-04-08 by waterlubber

sthalik, 2019-04-09:
Like I said do `WINEPREFIX=$HOME/foo opentrack`. Should there be more functionality for this?