opentrack/opentrack #933: Pitch axis limited to 90 degrees.

SoloWingX, 2019-04-23:
Hi. I am on OpenTrack 2.3.10.

I am using OT with my EDTracker Pro. I have a problem with the pitch axis. Looking up is limited to 90 degrees even though the output is set to 180 degrees and the octopus turns around fully. I remember that TIR5 didn't have such a limitation. Is there any workaround to allow more pitch in game? Like looking fully up and then some? It would help a lot in dogfights...

Tested with Il-2 CoD & DCS World.

Len62, 2019-04-23:
I don't think there is a way. You need the 180 output to get 90 up/down.

SoloWingX, 2019-04-23:
Could editing the configuration file lead to some results?

e.g pitch-max-value=180 to 220 or something to get the desired effect?

sthalik, 2019-04-23:
Looks like these games don't allow for pitch output over 90. See `doc/contrib/trackir-client` for a testing program.

closed 2020-01-27 by sthalik